FormerLurker/ArcWelderPlugin #156: Arc issues

Fabrizio3, 2021-02-11:
im using the cura plugin to start with, im trying octoprint one now, while it makes arcs in the gcode, i get 12 sided polygons instead of circles (around 9mm in diameter) and the slow af gyroid infill isnt much faster (im using klipper firmware btw, and arcs are enabled)
any idea why? maybe its the cura plugin will update soon, but i really wanted to use it as it would make circles and those dreaded SLOOW gyroid infills fast again (i print at 250mm/s with 2-3K accelerations so its pretty important to me)

FormerLurker, 2021-02-11:
First, make sure you are using the latest klipper firmware.  Second, find this setting:

```
[gcode_arcs]
resolution: 1.0
#   An arc will be split into segments. Each segment's length will
#   equal the resolution in mm set above. Lower values will produce a
#   finer arc, but also more work for your machine. Arcs smaller than
#   the configured value will become straight lines. The default is
#   1mm.
```
and change it to 0.1.  1.0 is WAY too large.  Some ppl even use 0.01, but that seems insanely small to me :)  I guess your mileage may vary.

Fabrizio3, 2021-02-11:
hi! thank you very much! didn't expect such a quick response, yeah the octoprint plugin is way better! cura was weird for some reason, maybe my settings, anyways works super well! arcs sound butter smooth now

closed 2021-02-11 by Fabrizio3

Fabrizio3, 2021-02-11:
maybe this will help someone in the future struggling with cura, and good to know about that 0.1mm!  your plugin is awesome you wouldn't believe how fast it goes on curves!

FormerLurker, 2021-02-11:
@Fabrizio3, 

> you wouldn't believe how fast it goes on curves

I'm building a printer right this second to test it out.  It's designed for development (swapping boards, motors, experimentation, etc..) and speed.  I want to see what the real differences between arcs and segments in high performance situations on different hardware/software...  I can't wait to join the club :)

Also, thanks for your compliments!