FormerLurker/ArcWelderPlugin #157: Octoprint realtime printing statistics are wildly off

chschmit, 2021-02-11:
The printing layers and percentage complete are off quite a bit. As is the realtime Gcode viewer.

I printed a fairly simple model with 100 layers and it showed:

Layer: 25/100
25% complete

When in actuality the print was about 95% complete with only a few hours left to go.

The Gcode viewer is also lagging behind and doesn't match what is being printed even when the 'sync with job progress' is checked.

Is OctoPrint unable to process G2/G3 codes for printing estimates?

FormerLurker, 2021-02-11:
> Is OctoPrint unable to process G2/G3 codes for printing estimates?

I'm not sure, but I'd guess Octoprint or whatever progress plugin (print time genius?) you are using isn't handling G2/G3 for whatever reason.  Anyway, this is a downstream problem, so it should probably go in either the OctoPrint github issues or whatever print time estimator plugin you are using.

Also, if you aren't using the most recent version of OctoPrint, you should update.  Some fixes for G2/G3 have been recently added.  Support for those commands in other apps (slicers, visualizers, estimation, servers ect...) is lagging a bit behind since people are only recently using G2/G3 regularly (probably due to ArcWelder).  I would expect this situation to be better in the coming months and years.  In fact, I've personally submitted several pull requests and issues trying to get better G2/G3 support.

chschmit, 2021-02-11:
Thanks. I'm on OctoPrint 1.5.3 and the print statistics are coming from within the "State" section in Octoprint. I'm not using any other progress plugin. I'll do a bit more investigation. 

I did find that if I send a file to OctoPrint, it converts using AW fine and the prints it, the printer will complete the print, but OctoPrint may not show the file as being completed. If I resend the same file from my slicer OctoPrint says there is a filename clash. I have to Cancel the print (that actually completed) before I can send a new version of the file for it to overwrite. I'm using PrusaSlicer and this seems to be a new anomaly since adding AW.

FormerLurker, 2021-02-11:
Hmm, I've not experienced that.  Are you uploading from PrusaSlicer, or using an 'Upload and Print'?  Also, you might consider (if you aren't already) [upgrading to the maintenance release candidate of ArcWelder](https://github.com/FormerLurker/ArcWelderPlugin/discussions/141).  The plugin itself (not the arc code) is practically a complete rewrite.  Lots of fancy new features and bugfixes.

chschmit, 2021-02-11:
I'm using the PrusaSlicer's Export and Print feature...one button.

I'll give the RC of AW a shot. Thanks!

FormerLurker, 2021-02-11:
You have a few options for dealing with the PrusaSlicer thing.  The easiest is to just [weld directly from prusa slicer](https://github.com/FormerLurker/ArcWelderLib#running-arcwelder-from-slic3r-slic3rpe-prusaslicer-and-superslicer) when you want to 'Export and Print'.  The new Arc Welder plugin will ignore any previously welded files, so you can still use the plugin when you want to.

KenLucke, 2021-02-13:
I'm definitely NOT seeing this, and I'm running 24/7.  I do use PrintTimeGenius, so that might be the difference.

craigwmiller, 2021-02-21:
Using PrintTimeGenius here, and can confirm time estimates look good here with Arc-Welder.