FormerLurker/ArcWelderPlugin #158: plugin cutting corners (Literally)

J-Phi1123, 2021-02-13:
Arc Welder turns the corners it straight lines. 
[Block Spacer_single.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/5974850/Block.Spacer_single.zip)

FormerLurker, 2021-03-01:
Sorry it's taken me so long to get back to you....  I've been so backed up lately that lots of things have fallen through the cracks, so I apologize.

I took a look at your arc welded gcode and this is what Simplify sees:

[image]

If corners are coming out flat, there are two possibilities:

1. Your printer has arcs disabled.  This means large and small arcs will not print correctly.  Try sending a ```G2``` command with no parameters and see if your printer responds with 'Unknown Command' (arcs disabled), or 'Invalid Parameters' (arcs enabled).
2. The print you are showing has quite small arcs, so if you are running a fork of Marlin 1.0 (or many other firmwares without mature G2/G3 commands) you may run into flat edges for small arcs.  if so, you can try upgrading to the latest beta version of arcwelder, where I add 'Firmware Compensation' to deal with this limitation, though it will reduce the compression for small arcs. 

Let me know about your printer's response to #1, and if you want to try #2.

J-Phi1123, 2021-03-02:
Nice, thanks for getting back to me.

1. G2 is working because I've used arcwelder with great success.   It has
sped up and smoothed out edges for a bunch of my prints,  well done.

2.  I am running latest kipper, the version escapes me, but I updated it
about 2 weeks ago,  and I am using the latest arcwelder octoprint plugin to
optimize the gcode.

The photo I was showing was how the gcode was being rendered in a gcode
viewer before and after arcwelder optimized the curves.

J-Phi1123, 2021-03-02:
Re read what you sent.  I uploaded two sets of gcode, before and after.
That is the before gcode.

FormerLurker, 2021-03-02:
> The photo I was showing was how the gcode was being rendered in a gcode
viewer before and after arcwelder optimized the curves.

@J-Phi1123, email attachments don't get added to the issues, if you have any photos to upload you'll need to do that via the github website.  If you already did that, something bad must have happened :)  Also, many gcode viewers (cura, Slic3r, Prusa Slicer, Super Slicer, etc...) can't render G2/G3 commands properly.  There are open issues for this in most of the repositories, so hopefully it gets fixed.  So far only Simplify3D can render them properly (albeit at a fairly low resolution).

> Re read what you sent.  I uploaded two sets of gcode, before and after.  That is the before gcode.

Are you talking about the photo I uploaded?  That is the after gcode.  I'll show both this time (notice the one of the right, the arc welded version, has far fewer movements):
 
[image]

> I am running latest kipper, the version escapes me, but I updated it about 2 weeks ago

You will want to change the mm_per_arc_segment setting from 1.0 to maybe 0.1 (or lower if you can get away with it, but start at 0.1) so that small arcs render correctly.

I'm guessing you haven't tried to print this gcode file?  If not, try it after changing the arc settings I mentioned.  If you have, take a pic of it please.