FormerLurker/ArcWelderPlugin #161: Immediately Load Processed Gcode and Start Print

laebshade, 2021-02-14:
As an Octoprint and Arc Welder user, I would like Arc Welder to immediately load processed gcode and start printing.

Use case: I use SuperSlicer, a PrusaSlic3r fork, to send gcode to Octoprint. The Octoprint integration has an option to immediately print; however, this means the file printed is the original gcode, not the Arc Welder modified. A workaround is to uncheck the print option; however, that means I have to open Octoprint and click the gcode processed by Arc Welder to load and print.

laebshade, 2021-02-14:
I wanted to open this in case I missed something or you're already working on it.

I'd be open to working on this; my day job is a software developer; I have several years of Python experience.

laebshade, 2021-02-14:
Leaving a breadcrumb:

The [REST API /api/job endpoint](https://docs.octoprint.org/en/master/api/job.html#issue-a-job-command) will allow job commands to be issued, most notably `start`. The file has to be selected first using the [/api/files endpoint](https://docs.octoprint.org/en/master/api/files.html#sec-api-fileops-filecommand).

Developing plugin docs: https://docs.octoprint.org/en/master/plugins/index.html

Open questions:

1. Can you make calls to the REST API from a plugin?
2. Is there a Python API for Octoprint for working with the REST API?
3. Or is there some built-in but undocumented (or I can't find it) feature for loading files and starting jobs?

laebshade, 2021-02-14:
Read your docs and found your recommendation to call the ArcWelder bin from SuperSlic3r: https://github.com/FormerLurker/ArcWelderLib#running-arcwelder-from-slic3r-slic3rpe-prusaslicer-and-superslicer

Since I can do the slicing with SuperSlicer, I no longer see any value in this open issue, so I'll close.

Thanks!

closed 2021-02-14 by laebshade

FormerLurker, 2021-02-14:
Hi @laebshade!  Glad you found the docs.  However, you may be interested in trying out the beta arc welder OctoPrint plugin, which contains the option you requested here.  See [this discussion](https://github.com/FormerLurker/ArcWelderPlugin/discussions/141).

laebshade, 2021-02-14:
Thanks for the reply.

I'll give that a whirl!