FormerLurker/ArcWelderPlugin #162: AW/Meatpack interaction

KenLucke, 2021-02-19:
Might want to take a look at this issue I just opened over in the MeatPack plugin:

https://github.com/scottmudge/OctoPrint-MeatPack/issues/16#issue-812324455

Looks as if it's MeatPack's fault (disabling it cures the issue), but it only happened on an ArcWelded file (and NOT on the same file before being welded)

FormerLurker, 2021-02-19:
Try disabling the space removal within meatpack.  Not all printers will support that.

KenLucke, 2021-02-19:
Will do, but it's been functioning fine on all other models for over two weeks.

KenLucke, 2021-02-19:
I take it back, this one is same dimensions, but the layer height was changed from 0.2 to 0.24.  That's the only difference in the model (which should not make any difference at all for this) from the one that previously printed properly.

KenLucke, 2021-02-20:
> Try disabling the space removal within meatpack. Not all printers will support that.

Nope, no change with the same file, same printer, same firmware.  Still does not print.  It's ONLY this one file, so far.

FormerLurker, 2021-02-21:
OK, it was worth a shot.

KenLucke, 2021-02-21:
> OK, it was worth a shot.

Input is always welcome, even when it doesn't work out. :)

jgason, 2021-02-27:
For what it is worth I can completely confirm the problem. Same here on a simple xyz cube. The non arc welder version works, the arc welder version do load but, for the lack of a better description, « phantom prints ». Octoprint and the Octoprint display plugin display layer advancement, the gcode viewer of octoprint does display progress of the nozzle, but not a blink happen on the real printer. Nothing, nada, que dalle... really strange.

FormerLurker, 2021-02-28:
So @jgason, would you say the original gcode with meatpacker works, the arcwelded version without meatpacker works, but not both together?