FormerLurker/ArcWelderPlugin #163: Cannot Install ArcWelder. Stuck at "Building wheel for Arc-Welder"

CharlesPavlov, 2021-02-20:
ArcWelder becomee stuck at "Building wheel for Arc-Welder" when installing from the plugin manager. I've been having this issue since the end of 2020. Never got to install ArcWelder. I'm at my wits' end. Please help...

Summary:
Octoprint Version 1.5.3 running on Raspberry Pi Mk3B
Octopi 0.17
Python 3.7.3
PIP 21.0.1
Attached screenshot, octoprint.log and plugin_pluginmanager_console.log and Octoprint System Information.

System Information:

> browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:85.0) Gecko/20100101 Firefox/85.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1400
env.hardware.ram: 904646656
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x80000
env.python.pip: 21.0.1
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.3
[image]
[plugin_pluginmanager_console.log](https://github.com/FormerLurker/ArcWelderPlugin/files/6015518/plugin_pluginmanager_console.log)
[octoprint.log](https://github.com/FormerLurker/ArcWelderPlugin/files/6015519/octoprint.log)

FormerLurker, 2021-02-21:
Hmm..  It can take 5 minutes or more to build on a pi3B, but I'm guessing you've let it go longer than that.  If you have (again, you probably did), I think we can try 2 things:

1. Enable verbose logging for PIP
2. Install manually from the command line.

The problem with option 1 is I've forgotten how to do it.  It's some parameter that needs to be added in the plugin manager settings.  Option 2 takes some instructions, which I'll try to put together.

Just let me know if you've waited for a good long while for the install, and I'll put together the steps for the other options I listed.

CharlesPavlov, 2021-02-21:
I've let it run overnight, a couple of times... As of this moment, it's been running for a good 6 hours. I'll stop it and see if I can convince PIP to spew out more logs, which I'll post as soon as possible.

FormerLurker, 2021-02-21:
I figured..  if I don't hear back by tomorrow, i will start digging up that parameter for the plugin manager as well as manual installation instructions.

CharlesPavlov, 2021-02-21:
Managed to get the plugin manager to run PIP with the "-v" option. Ran it for around 45 minutes. Output below.

[plugin_pluginmanager_console.log](https://github.com/FormerLurker/ArcWelderPlugin/files/6018135/plugin_pluginmanager_console.log)
[octoprint.log](https://github.com/FormerLurker/ArcWelderPlugin/files/6018136/octoprint.log)

FormerLurker, 2021-02-21:
Have you tried updating your OS via ```sudo apt update``` and ```sudo apt full-upgrade```?  I checked the c++ compiler version, and you have the same one I have installed.. Hmm..  Are you having any PSU problems on your pi (look for the lightning bolt)?  Maybe you could run HTOP or something while it is installing and see if it's a memory issue?  

Alternatively, try installing arcwelder via a terminal like so:

```
~/oprint/bin/pip install "https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip"
```

I'll continue to look over the verbose pip output, but it just seems to die for some reason without any errors....

CharlesPavlov, 2021-02-21:
I'm running my Pi through a SSH terminal so I don't think I would see the lightning bolt icon (any other way to check?). Never seemed to have any obvious stability issues though...
I'll do a 'apt full-upgrade', then do a manual install then get back to you.

FormerLurker, 2021-02-21:
Check this link: https://community.octoprint.org/t/pi-undervoltage-power-issues-i-had-a-strange-lightning-bolt-temperature-symbol-with-an-exclamation-mark-popup-in-my-navbar-what-does-that-mean/5256?page=2

CharlesPavlov, 2021-02-21:
Ran `apt full-upgrade`, running `~/oprint/bin/pip install "https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip"` right now. Looking into possible power issues too, but a summary grepping of syslog for "voltage" (`cat /var/log/syslog | grep voltage`) returns nothing.
`vcgencmd get_throttled` returns `throttled=0x80000`
If `0x80000` converts to `1000 0000 0000 0000 0000` does it mean that, according to https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md , that there is no power related throttling going on?

FormerLurker, 2021-02-22:
I believe 0x80000 means you are encountering thermal throttling.  Most install issues I see (both for ArcWelder and octolapse) are related to power/temp throttling.  I'm not sure why exactly it interferes so much, but it definitely does.  I rarely see thermal throttling, but it does happen.  Maybe grab a heatsink and aim a fan at the rpi until installation is completed?  i find passive cooling to be adequate for my pi, but it depends on many factors.  My pi4 requires active cooling else I run into issues.

CharlesPavlov, 2021-02-22:
Well, running it caseless out the window on a cold winter night. Must be around 0-2 Celsius outside. Fingers crossed :D 
Been running for around ten minutes now.

FormerLurker, 2021-02-22:
> By the way, is there any difference between running ~/oprint/bin/pip install "https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip" -v and installing it from the plugin manager?

well, just that it isn't running through OctoPrint.  There is less to go wrong.  You could try booting up OctoPrint in safe mode, then try installing from the terminal.

Also, this is such an odd problem.  Not sure why this happens occasionally.  If you happen to have an extra SD around, you might try re-flashing OctoPi as a test, and see if it installs from a clean flash.  Doesn't really solve the issue, but it might yield some info.

CharlesPavlov, 2021-02-22:
This is it at around the thirty minute mark.
It did get further than last time (it used to hang at `2021-02-21 17:48:17,750 ! GGC heuristics: --param ggc-min-expand=88 --param ggc-min-heapsize=110430
2021-02-21 17:48:17,750 ! Compiler executable checksum: 7defdc925cf5fede452fc531d54623d1`).

I might give it a couple of more minutes then if it's still not past this point I'll boot into safe mode and try again.

FormerLurker, 2021-02-22:
If it's progressing, I'd leave it alone!  I LOVE your cryo-cooling technique, btw, lol!

CharlesPavlov, 2021-02-22:
It hadn't progressed for about fifteen minutes I think. I killed it, started Octoprint in SafeMode, figuring it might save some system resources and then ran it from the terminal again.
It seems to be getting stuck in the same place as fifteen minutes ago...
If you find any use for it, I could also attach the terminal log...
[image]

CharlesPavlov, 2021-04-23:
FormerLurker, just wanted to let you know, ArcWelder installed successfully sometime after updating to OctoPrint 1.6.0rc2. Can't really tell what got it to work though...