FormerLurker/ArcWelderPlugin #165: Cannot fetch update data for arc welder

MrDragon420, 2021-02-24:
octoprint.plugins.softwareupdate.exceptions.ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None
2021-02-22 12:57:31,167 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2021-02-22 12:57:32,776 - octoprint.plugins.softwareupdate - ERROR - Could not check arc_welder for updates

cp2004, 2021-02-28:
Try installing the release candidate version, this has been fixed for a while now:

https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.1.0rc2

FormerLurker, 2021-03-01:
Thanks @cp2004, somehow this fell off my radar.  Too much going on :)  Sorry @MrDragon420 .

MrDragon420, 2021-03-01:
I do not see how to install this, i do not know what i am doing, can someone point me to what i need to do?

FormerLurker, 2021-03-01:
You're probably just using the wrong URL.  Did you look at [this](https://github.com/FormerLurker/ArcWelderPlugin/discussions/141) already?