FormerLurker/ArcWelderPlugin #169: Filled circles?

marcdraco, 2021-03-02:
Looks like a bug, whenever I try to process files with cut circles (for axels, bolts etc.) the STL file is file and the AW processed file looks correct in Octoprint BUT the printed file has what looks like stringing, only it's really, really thick, right across the centre of smaller holes.

Not sure if this is a problem with the iffy firmware or in AW itself.

Yours, 

Confused.

seantapscott, 2021-03-02:
Hey Marc,

I'm Sean, I just follow this thread and help out with replying if I can
sometimes.

Just popping to say that I have experienced this before, and when I have,
it was related to the "G90 Influences Extruder" setting.  What
printer/firmware version are you running?

marcdraco, 2021-03-02:
Hi Sean,

It's a TronXY Pro 2 with stock firmware - I can't put Marlin on there much as I'd like to. It's not that I'm unable, just that I don't want to jigger the warranty - mind you, I've already had to replace the entire hot end with something similar - but is far more reliable even though it's a clone.

I needed this to make it work properly (if that means anything to you):

https://github.com/OctoPrint/OctoPrint-FixCBDFirmware

seantapscott, 2021-03-03:
So, this is probably running low-feature Marlin, like version 1 or so.  Can
you run M503 and send us the results? That should give the marlin version
number as well.

marcdraco, 2021-03-03:
Sure will. It's just completing a print so I'll do that as soon as it completes. For the money this is a well-featured and fairly well made printer (except that infernal hotend and the odd wiring runs) but it's an absolute b'stard that I can't upgrade the firmware. I haven't checked the Marlin licence but most open source requires us to release our changes/fixes for using it. (My own is BSD 2-clause so but it's not the sort of thing someone would necessarily need/want to modify on a commercial level.)

marcdraco, 2021-03-03:
M503 just prints some numbers which I can only assume it's some non-Marlin or something broken beyond that... Foosel's fix notes as following during startup though if this helps.

Repl: M110 N0 -> M110
Send: N0 M110*35
Changing monitoring state from "Connecting" to "Operational"
Recv: ok N:0
Repl: M110 N0 -> M110
Send: N0 M110*35
Recv: ok N:0
Send: N1 M115*39
Recv: ok CBD make it.Date:Nov 21 2019 Time:16:32:55
Repl: ok CBD make it.Date:Nov 21 2019 Time:16:32:55

 -> ok CBD made it, foosel fixed it.Date:Nov 21 2019 Time:16:32:5

seantapscott, 2021-03-03:
Sorry, my original message was incorrect, I was upposed to say M115, I
edited it on github, but you probably already got the email with the
mistake.  M503 reports printer capabilities. M115 prints firmware info.

marcdraco, 2021-03-03:
Ah! No sweat Sean... on sec. On snap - just comes out with the same hopeless message (CBD, not Foosel!)

Send: N6 M115*32
Recv: ok CBD make it.Date:Nov 21 2019 Time:16:32:55
Repl: ok CBD make it.Date:Nov 21 2019 Time:16:32:55

 -> ok CBD made it, foosel fixed it.Date:Nov 21 2019 Time:16:32:55

I'm not familiar with Marlin other than knowing it exists - is this any clearer? I note that Arc Welder doesn't seem to have issues with larger holes - the 22mm diameter ones for the bearings I'm using are fine, it was the 6mm ones that got filled in,

FormerLurker, 2021-03-03:
Hey, thanks for all the help @seantapscott!

@marcdraco, it might be good if you took a pic of what you are seeing?  Also, before/after gcode would be good too, so I can make sure the code itself is reasonable.

marcdraco, 2021-03-03:
That's a LOT of g-code dude. How can I limit it to just the bits you need? I don't (currently) have any remaining prints showing that bug IRL as it were and I'm printing most of the day. I'll certainly run something off so you can see it though. (I've just been drilling out the holes previously since these are all "rapid" prototypes.

The original sliced STL comes from Cura.

I believe the error is with the TronXY because while the hardware is pretty decent (for the price) the software is ... well a bit hit and miss and now the damn thing has decided to cram itself into the bed... oh joy.

FormerLurker, 2021-03-03:
Post to gist.guthub.com and drop a link here.

seantapscott, 2021-03-03:
No worries, Marc.  Here's a reference link I always keep open to Marlin's
gcodes.  It is helpful for understanding what your gcode is doing, and
writing new filament loading scripts, start/end gcode, and other stuff.
https://marlinfw.org/docs/gcode/M115.html

In other news, that printer's firmware is... not great. Looks like it is
broken, when I read that link you mentioned earlier.  I did find some
options for upgrading the firmware like this link:
https://www.tronxyonline.com/Firmware-XY-Series-a63177.html, but that is
hard to recommend because I simply don't know if that's even official.
Looks like they're not exactly upgrading the firmware, or rewriting it, but
actually using a series of gcode commands to rewrite the configuration
settings written by the firmware?

marcdraco, 2021-03-03:
Yeah, that's my take too. I have Cura set to do an auto-bed level each time it prints *just in case* something went pear shaped overnight, etc.

And of course, it was just my luck that I caught one of the little clips holding the glass platter in place (another upgrade). But those clever folks at TronXY used a bed levelling sensor that picks up on stuff like that and ... of course it did which meant it corrected the bed level to be about 0.5mm lower than it actually was! Luckily I heard the extruder clicking so I caught it.

 I think we're going to have to put this one down to crappy FW since the previews produced in Octoprint (albeit based on Cura's older code) show the process is fine. It's only when I print smaller radii that it goes a bit weird.
 
I'm obviously going to have to dig deep enough to get something that actually RUNS Marlin but I've got used to auto-bed levelling now so that's a bit of a drag. Still, I prefer to be able to see the source code I'm running so that's another reason. I did speak to TronXY and Amazon about this. TronXY just said, "we strongly recommend against it although we know people have" and Amazon banned me from commenting any more. Gotta love those moderators - give us unbiased, expert reviews - unless you happen to be an unbiased expert and we hear something we don't like!

FormerLurker, 2021-03-03:
If you can, please do take a pic of the part that had issues with the small circles.  It might be a long shot, but I've seen just about everything, so maybe it's something that can be fixed.

marcdraco, 2021-03-03:
Of course. I'll send something I know will go "wrong" after the current print finishes. I'll do a couple of tests to see if cylinders are affected too - I've printed a few of those (bearing carriers) and they are printing fine so far but I might have missed something with all the juggling. I THINK the problem crops up when the small circle is surrounded by lots of infill and space.

Let's see! (Gawd I almost channelled by inner Dr. Vax then! I love Irv.)