FormerLurker/ArcWelderPlugin #170: ArcWwelder creates "Blobs" on in and outside of a circular object

Seri-, 2021-03-04:
I have been using ArcWelder for quite a whiel and this is the first accurence of such a blatant error. The top ring is printed from the original file the bottom from the created Arcwelder file.
I am running the latest dev. version of Arcwelder on a about one month old version of Marlin bugfix.
The used firmware as well as both gcode files can be found in the google drive.

[image]
[image]

https://drive.google.com/drive/folders/19Id1-YcuNoUH5U-bXOYu9U6d96RRy7V-?usp=sharing

Typho81, 2021-03-05:
Hi,
I have a different configuration (Marlin 2.0.7.2, stable version of ArcWelder 1.0, BTT SKR V1.3) and same issue.

FormerLurker, 2021-03-10:
FYI, still looking into this.  I pretty much have to work through the gcode by hand since no visualizers I know of handle flow rates.  I am adding some enhancements for Cura Arachne (flow rate detection) that should help debug this, so it's on the back burner until i have that enhancement ready.

FormerLurker, 2021-03-30:
So, I've gone over several layers by hand to make sure the extrusion lengths add up correctly, and so far I can't find anything wrong with the welded gcode.  I'm going to try printing the first couple layers myself, just have to alter the start gcode.

Seri-, 2021-03-30:
So it might "just" be marlin missinterpreting something or not handling the arcs correctly? I have printed the "welded" code two times and both had the same errors.

FormerLurker, 2021-03-30:
It's printing now, so we will see.  Your model BARELY fit on my build plate.  I had to remove the skirt.  So far it looks OK, but time will tell.  I'll let it print up a few MM so it will be obvious.

Sorry this took so long, but it is a pain to verify extrusion amounts :)

Seri-, 2021-03-30:
> 
> 
> All good, not like you owe me anything ^^. But urm, barelly fit? it is only like 10,7 cm in diameter. Did you accidentally translate the gcode to inces or something? :P

closed 2021-03-30 by Seri-

FormerLurker, 2021-03-30:
>  But urm, barelly fit?

It's not the size, but the position.  My bed goes from 0-210mm in the y direction, and the skirt crept over a bit.

So, the print is finished, and though i had some layer adhesion issues (Your print speeds are too fast for my printer, and cooling too low for my filament), I don't see any blobs.  Maybe try updating marlin and retry the print?

Seri-, 2021-03-30:
ah that makes sense. Will do but not today sorry :( gimme till tomorrow evening.

seantapscott, 2021-03-30:
I had this the other day because the printer mainboard was not processing
the commands quickly enough.  If you watch it, it probably pauses and hangs
for a second and oozes at each blob point.  I suggest reverting back to a
known good firmware.