FormerLurker/ArcWelderPlugin #171: [FR] Option to not convert filename spaces to underscores

lightmaster, 2021-03-07:
Can you add an option to not convert the filename spaces to underscores? PrusaSlicer Thumbnails reformats the State panel to show the filename next to a thumbnail, and underscores in a filename cause it to be all one "word" and too long. Also, not sure what the reason for the conversion of spaces to underscores since pretty much every system these days handles spaces just fine.

[image]

[image]

FormerLurker, 2021-03-10:
I believe I have fixed this in the latest pre-release versions.  Feel free to check it out under releases.  There is a page in the 'discussions' tab that gives installation instructions.

lightmaster, 2021-03-10:
I don't see any option about it and its definitely still automatically changes spaces to underscores.

FormerLurker, 2021-03-10:
So what version are you using?

lightmaster, 2021-03-10:
1.0.1 rc2

FormerLurker, 2021-03-10:
Ok, I'll take a look.  I have to say I'm not seeing this in my version, but I don't have the prusa thumbnail plugin installed at the moment.  The internal storage name is changed (includes underscores), but that's an OctoPrint thing, and is done for multi-platform support.  I'll dig in and see what I can discover.

lightmaster, 2021-03-11:
The only 2 plugins that I have that should have any affect on files or the file browser are UI Customizer and PrusaSlicer Thumbnails. I disabled both of them and uploaded a new file and still saw the spaces get converted into underscores. I tried both drag and drop and uploading directly through PrusaSlicer.

FormerLurker, 2021-03-11:
Now i am not sure what you are referring to.  The only way info you suppled about what file name you are talking about is the thumbnail plugin display.  Where else is this happening.  Please be as specific as possible.

lightmaster, 2021-03-11:
Once the file is converted to whatever.aw.gcode by AW, the filename shows up in the Files Browser with the underscores. The original file shows its filename with spaces, but the "internal filename" has the underscores.

[image]

About those plugins, I was just listing the only plugins that I have that should be able to modify the Files Browser at all.