FormerLurker/ArcWelderPlugin #175: ArcWelder seems to cause some odd moves away from the print while extruding

samwiseg0, 2021-03-11:
#### Description
Arc Welder seems to cause some odd moves away from the print while extruding. The issue manifests it self on random layers where the hotend continues to extrude, movies away from the print, moves to the max x and y and then circles back. It causes "strings" to be created and they connect back to the print where the hotend eventually resumes.  

This issue does not seem to be just me as there is a reddit post that has several users including me experiencing this.

[Here](https://www.reddit.com/r/ender5/comments/m2nu78/printer_did_a_corner_tour_mid_print_why/) is the post from reddit including pictures of what happens to the print. 

#### Version

**Arc-Welder (1.1.0rc2)**

Marlin Bugfix 2.x.x commit [fd455be55cca886d5518ff7efeaee007802abbe1](https://github.com/MarlinFirmware/Marlin/commit/fd455be55cca886d5518ff7efeaee007802abbe1)

gargamel778, 2021-03-11:
[CE5_BlackbirdSR71_P1.gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/6124606/CE5_BlackbirdSR71_P1.gcode.zip)

Gcode attached.
Gap can be observerd at ~87.7mm.  I've noticed this happend also on lower layers, leaving 2 strings and a zit in the middle.

[image]

ldursw, 2021-03-25:
This looks similar to #30.

If you are using OctoPrint keep an eye on the Resend Ratio. It should be always zero. If it is not then the command is not being received correctly by the printer and may cause artifacts. There are some things you can try.
- Printing from SD card. If the issue doesn't happen then it is something to do with serial communication.
- Testing the latest Marlin bugfix with `RX_BUFFER_SIZE` set to `128`. Arc commands are longer and the buffer size may need to be increased.

samwiseg0, 2021-03-25:
> Testing the latest Marlin bugfix with RX_BUFFER_SIZE set to 128. Arc commands are longer and the buffer size may need to be increased.

I have not seen that recommendation before. I can try that. I have not noticed there has ever been anything being resent in octoprint. I usually look at that value and I have never noticed any resends.

Also based off how the print head moves and off the object, while it extrudes, it does not seem like a resend? Wouldn't the printer just pause in place? This behavior has the print head moving the whole time but it just moves away from the print and circles the bed. Then moves back to the object.

I will see if I can replicate this issue with a test print first before changing anything.

Thanks for looking into this.

seantapscott, 2021-03-25:
Hey Samwise,

I've been watching the same thing on my Ender 5 Pro with the v4.2.2 board,
and I think it is actually a broken command being sent. Repetier server is
actually logging some of the resends and saying "Invalid command", and
it'll be the right command, but the first 7-8 ASCII characters will be
missing.  It is possible that occasionally it strips the line number off in
a way that makes sense to the printer.  I'd like to know why my one printer
does this action, but my other Ender 5 with an older Melzi board does not,
it just says "checksum failed, resend".  I've been trying to find a marlin
firmware setting that controls Checksum verification, but haven't found any
yet.

Anyway, I'm going to play with RX_BUFFER_SIZE too.

A fellow peer,
Sean

samwiseg0, 2021-03-25:
Interesting, it does seem the common denominator is the 4.2.2 board from creality here.

Let us know what you find.

I will try to arc a benchy and some other shorter prints to see if I can see this behavior and then make the changes to marlin. 

Fee free to upload any gcode you may have that exhibits this behavior. Preferably something that does not take too long to print for testing.

seantapscott, 2021-03-25:
I can test serial communications and the percentage of resends using
Repetierserver, and the problems occur when I ramp up to the 255 Input
Buffer Size.  But if I leave it at 127, it also complains that some of the
arc commands are too long for the buffer.

Anyway, the percentage of resends is between 0.5 - 1.0%, so I think a
longer file is necessary to truly see the action at play. And not all of my
resends are generating the gaps and Max-X-Max-Y movements, so there must be
a percentage of those 1%ers that are actually being interpreted incorrectly
by the board.

Hypothesizing only... I intend to recompile the firmware after the next
print job and I'll test.  I'm using TH3dstudios firmware.

seantapscott, 2021-03-25:
Raised my RX_BUFFER_SIZE to 256 and I was able to eliminate resends with
the Melzi board. I'll try it with the 4.2.2 board when it is done.  The
setting is in the configuration_adv.h file.

samwiseg0, 2021-03-25:
> Raised my RX_BUFFER_SIZE to 256 and I was able to eliminate resends with the Melzi board. I'll try it with the 4.2.2 board when it is done. The setting is in the configuration_adv.h file. 

Do you have a gcode file you know causes the weird moves? Not sure we can definitively say that `RX_BUFFER_SIZE` is the culprit without having a source file we are sure causes the issue?

seantapscott, 2021-03-25:
Agree, but I have other priorities now.  Definitely need further testing.

ldursw, 2021-03-26:
> Also based off how the print head moves and off the object, while it extrudes, it does not seem like a resend? Wouldn't the printer just pause in place?

> And not all of my resends are generating the gaps and Max-X-Max-Y movements, so there must be a percentage of those 1%ers that are actually being interpreted incorrectly by the board.

I've documented in more details in the issue I've linked. I'll try to explain what is happening.

Let's take the command `G2 X102.538 Y122.280 I-2358.205 J4442.201 E2277.49416 F1800` which has 59 characters. OctoPrint adds a counter and a checksum resulting in `N328763 G2 X102.538 Y122.280 I-2358.205 J4442.201 E2277.49416 F1800*58` which has 70 characters. The default buffer size is 64 bytes for the Creality 4.2.2 and similar boards.

As the command is larger than the buffer the oldest data is overwritten, causing Marlin to receive `3 G2 X102.538 Y122.280 I-2358.205 J4442.201 E2277.49416 F1800*58` (note the missing `N32876` at the start) and responding with `echo:Unknown command: "3 G2 X102.538 Y122.280 I-2358.205 J4442.201 E2277.49416 F1800"`. OctoPrint ignores this error and moves on.

Then OctoPrint sends another command, let's say `N328764 G1 X95.77 Y121.167 E2289.37068*99`. The firmware will reply with `Error:Line Number is not Last Line Number+1, Last Line: 328763`. The last command was not performed because it wasn't received correctly. OctoPrint sends the previous command again and if you are unlucky enough to have the command corrupted to just `G2 X102.538 ...` (all the `N328763` is gone) then the firmware will skip the command queue and perform the command immediately. The nozzle won't be in the location that was supposed to be and will do a circle around the build area. When the corrupted command finishes the print will continue as usual.

In short: You have to have 2 corrupted commands in a row to trigger the issue. The exact same gcode can have 0 or 100 resends or anything in between due to how OctoPrint sends the commands. If you can eliminate the resends then you solve the issue. If you have more than zero resends then it could have caused the artifact (but it may not because the command corrupted only once).

Also, this is not an issue with OctoPrint, any software that communicates with the printer via serial can cause this.

> Do you have a gcode file you know causes the weird moves?

Anything with a lot of arcs is a good candidate. To avoid using filament you can slice with the hotend and bed temperature set to 0 and print. It will do all the moves but won't extrude. Don't use the Dry Run plugin because it will reduce the command size and won't cause resends. [This model](https://www.thingiverse.com/thing:4636296) with gyroid infill will certainly cause resends with the default configuration.

> I'm using TH3dstudios firmware.

I don't know if their firmware has the latest (as in after February 27th) patches. The commit MarlinFirmware/Marlin@0cc03f9 is necessary to fix the issue otherwise the `RX_BUFFER_SIZE` configuration won't have effect.

seantapscott, 2021-03-26:
So arcs are causing this because their commands are longer and there is
currently no way to fix it outside of modifying or including the two files
that were rewritten inside that commit you referenced?

ldursw, 2021-03-26:
As this is a firmware issue the only way to properly fix it is to patch the firmware. An alternative is to print via SD card or not use ArcWelder at all but I'm sure you don't want that.

If it's possible try the latest bugfix from the official Marlin repository. If it's not then you will have to patch the code yourself, wait for TH3D to update their fork, or ask them to apply the patch.

samwiseg0, 2021-03-26:
@ldursw Thank you for the detailed explanation. That makes complete sense. 

My goal currently, is to reproduce the issue. So far I have done 3 prints and no issues. Once I have a gcode that produces the issue. I will review the serial log and look for the printer responses you mentioned. I am actually printing objects as it is easier to see the issue if it happens and I don't have to watch my camera in case I miss it. I will slice and print the STL you suggested next. 

After reproducing, I will pull the new bugfix firmware with the `RX_BUFFER_SIZE` modified and no other changes and run then test agin.

ldursw, 2021-03-26:
I'd say 98% of the prints won't have the artifacts but the underlying issue is still there. If the print finishes with anything other than zero resends then you reproduced the issue.

[image]

To make it easier to find resends you can enable the serial logging at `Settings -> Serial Connection -> Serial logging -> Log communication to serial.log` and restart OctoPrint. Then after printing download the log and search for "Resend".

samwiseg0, 2021-03-26:
> I'd say 98% of the prints won't have the artifacts but the underlying issue is still there. If the print finishes with anything other than zero resends then you reproduced the issue.
>
> [image]
>
> To [...]

Yep. I have that already setup.

Thanks!

samwiseg0, 2021-03-28:
> I've documented in more details in the issue I've linked. I'll try to explain what is happening.
>
> Let's take the command `G2 X102.538 Y122.280 I-2358.205 J4442.201 E2277.49416 F1800` which has 59 [...]

I can 100% confirm this was the issue. I wrote a script to check all my old gcode and found a few candidates that had greater than 64 bytes, including serial overhead, in the command. I then printed the file and waited for several resends. At one point it one of them did take a tour around the build plate as it was a corrupted command. I then stopped the print. Uploaded the new marlin bugfix with a higher RX value and sent the print to start again. I waited till it passed the point where the first print had issues. The second print had 0 issues.

Bellow is an example as @ldursw described of a corrupted command because of a buffer overflow. 

A couple things to note. I was unable to produce the longer strings with gcode sliced with PrusaSlicer. All of my "bad" files were sliced in cura. I am not sure if cura is doing something different than PrusaSlicer that could cause these longer commands to be generated.  

Would it be too much to ask if something could be included in arcwelder to check the gcode generated and warn the user that some of the commands generated will require higher RX buffers in marlin? Is there a way to ask marlin what the current RX buffer is and have that as part of the firmware check?

```
2021-03-27 13:50:32,158 - Send: N4766 G1 X94.484 Y97.282 E2402.55221*103
2021-03-27 13:50:32,173 - Recv: ok
2021-03-27 13:50:32,191 - Send: N4767 G3 X52.484 Y55.278 I163332.556 J-163359.002 E2404.77136*65
2021-03-27 13:50:32,209 - Recv: echo:Unknown command: "767 G3 X52.484 Y55.278 I163332.556 J-163359.002 E2404.77136"
2021-03-27 13:50:32,212 - Recv: ok
2021-03-27 13:50:32,217 - Send: N4768 G0 F9000 X52.643 Y55.119*117
2021-03-27 13:50:32,246 - Recv: Error:Line Number is not Last Line Number+1, Last Line: 4766
2021-03-27 13:50:32,251 - Recv: Resend: 4767
2021-03-27 13:50:32,258 - Recv: ok
2021-03-27 13:50:32,264 - Send: N4767 G3 X52.484 Y55.278 I163332.556 J-163359.002 E2404.77136*65
2021-03-27 13:50:32,595 - Recv:  T:209.80 /210.00 B:55.00 /55.00 @:83 B@:30
2021-03-27 13:50:34,275 - Recv: echo:busy: processing
2021-03-27 13:50:34,590 - Recv:  T:210.00 /210.00 B:55.01 /55.00 @:81 B@:27
2021-03-27 13:50:35,118 - Recv: ok
```

Thank you @ldursw for all your help and pointing me to the solution. I hope this helps others.

seantapscott, 2021-03-28:
I just compiled from the latest marlin bugfix (which was fairly painful),
and I can confirm.  I set the Input Buffer Size in Repetier to match the
firmware's new receive buffers at 1024 bytes, and I'm now pushing 216.7
lines/s, 250000 baud rate with no resends. I also enabled full circle arcs,
although I don't think arcwelder supports that yet.

Sean

FormerLurker, 2021-03-28:
> I also enabled full circle arcs,  although I don't think arcwelder supports that yet.

You are correct:  Arc welder won't generate full circles.  However, I've never seen any gcode from any slicer that would produce a full circle anyway.  Toolpaths rarely overlap in that way (it would create artifacts).

Also, amazing work in this thread.  I don't think I would have ever figured this out myself!

seantapscott, 2021-03-28:
Agreed on full circles possibly being unneccessary. I do have a strawberry
planter that might actually be close to using them. I design and print
planters as a business.

[image: image.png]

It is interesting to watch, as the light diffuses off of the surface
differently now with arc welder.  But where the indents are, the arcs do
wrap nearly all the way around, and then switch to an inner wall.  There
really isn't much benefit to implementing full circles though.  A longer,
slower command might just introduce more problems than fix.

FormerLurker, 2021-03-28:
@seantapscott, if you encounter any gcode that has a full circle (starting XY = ending XY) send it my way and I'll take a look.  I'm totally not opposed to implementing this, just haven't come across it yet.

I'm finally getting back to cleaning up some issues today (been totally immersed in other projects, mostly building a new printer to speed up testing), so it's a good time.

seantapscott, 2021-03-28:
Clarification question.

What does a complete circle look like in gcode?  I see that the X and Y
coordinates are the destination, but how would it look?

Like, I think this is close, but I'm not sure:
```
G3 X85.034 Y91.201 I21.045 J-20.768 E48.5724
M204 S500
G0 F7500 X84.743 Y90.927
M204 S400
;TYPE:WALL-OUTER
G3 X83.835 Y91.98 I22.25 J20.082 E55.56629 F1200
```
Also, should we open a separate issue for this?

FormerLurker, 2021-03-28:
It's tricky to create a G2/G3 that is a full circle.  However, finding the G0/G1 commands to generate one aren't too difficult.  I could hand create one, but I'd like to see this come out of a slicer.

Feel free to create another issue!  It will probably be a back burner issue, though, unless I can find slicers outputting full circles via G0/G1, so keep that in mind.

closed 2021-03-28 by FormerLurker

crsnwby, 2023-01-07:
Removed it, was causing all prints to mess up.