FormerLurker/ArcWelderPlugin #176: Complex modle crashes Pi at 45% processed by AW

GhostlyCrowd, 2021-03-13:
I may be reaching the hardware limitations of the pi? I'm not sure, but the Pi hard crashes, as in dead, no ssh nothing left it for 20 minutes it never recovered it was dead. 

Hard reboot fixed it. The same model I can AW via my slicer. But I'm unsure if it's as good as doing it on the Pi (do they both achieve the same welding accuracy with default settings?  

Attached is the Original  gcode via link as its too large for github https://1drv.ms/u/s!Aht5eQ6fqQw3ofVh9WYgtIbxYxckLQ?e=yGDTvh

FormerLurker, 2021-03-15:
What kind of pi do you have?  I've only ran arcwelder on Pi3B+ and Pi4, but I do know ppl who have used a Pi2 successfully, though slowly.  I think a Pi Zero would take forever, and a Pi1 would be even worse.

GhostlyCrowd, 2021-03-15:
> What kind of pi do you have? I've only ran arcwelder on Pi3B+ and Pi4, but I do know ppl who have used a Pi2 successfully, though slowly. I think a Pi Zero would take forever, and a Pi1 would be even worse.

Pi 3b+ with cooling and have allowed it to use the full clock rate of the SoC 1.4GHZ so i was surprised when it locked up and crashed the pi

FormerLurker, 2021-03-15:
Can you try something for me:  disable all third party plugins you have installed, reboot, and try welding again?  Sounds like it could be some kind of conflict.  If not I'll have to dig in a bit more.

GhostlyCrowd, 2021-03-15:
> Can you try something for me: disable all third party plugins you have installed, reboot, and try welding again? Sounds like it could be some kind of conflict. If not I'll have to dig in a bit more.
Sure
I will do this when my current print finishes.

GhostlyCrowd, 2021-03-18:
I tried to recreate this and now its fine, so guess it was a one off.

closed 2021-03-18 by GhostlyCrowd

FormerLurker, 2021-03-18:
That is so odd!  Well, I'll keep my eyes/ears peeled for any similar issues while testing.  Thanks anyway for your report!