FormerLurker/ArcWelderPlugin #177: Removes thumbnail

Daskew78, 2021-03-13:
Hi,

I've been scratching my head trying to work out why my thumbnails generated by Cura were not working with Octodash. 

It appears ArcWelder removes this when processing the file.

Is there any way I can stop it removing the thumbnail?

Thanks for your help.

primeinc, 2021-03-15:
This was fixed in 103ad7d63de2c2cc76803ef7ad1e7a58abd79df1 which I believe should be apart of [1.1.0rc2](https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.1.0rc2)

Daskew78, 2021-03-16:
Sorry for the delay, just been very busy...

I've re-installed and it's reporting my version is: (1.0.0+u.bb71e8f)

Sounds like I have an older version, how can I install a more recent version please? I'm simply searching the plugins within Octoprint and installing it that way.

Thanks.

seantapscott, 2021-03-17:
Hi Daskew78,

Looks like there are instructions for changing your release channel for
Octoprint in the Readme file on ArcWelder's page.
https://github.com/FormerLurker/ArcWelderPlugin

Sean

Daskew78, 2021-03-17:
> Hi Daskew78, Looks like there are instructions for changing your release channel for Octoprint in the Readme file on ArcWelder's page. https://github.com/FormerLurker/ArcWelderPlugin Sean
> […](#)

Thanks, i'm new to Github so didn't notice that. I've managed to install the latest version now. Cheers.

seantapscott, 2021-03-17:
No worries, we're all new at some times.  FormerLurker has put together a
beautiful readme page, and updates it regularly.

Daskew78, 2021-03-17:
I'll keep an eye on it, thanks again!

FormerLurker, 2021-03-17:
@seantapscott, you are a hero!  Thank you for your continued contributions, they are super helpful and appreciated!

seantapscott, 2021-03-17:
They are a small help, but I hope you see my recognition of your hard work
and my good intention of giving you more time.  First level support, if I
can provide it.

soostdijck, 2021-07-06:
Came here looking at the same issue (for me is seems to sometimes delete the thumbnail but now always). Sadly installing the RC with the fix is blocked by #174 

Here's to hoping for a real release soon!

FormerLurker, 2021-07-06:
@soostdijck, can easily install by using the *from url...* box in the plugin manager.  Let me know if you're interested since I'll be looking for testers soon.

soostdijck, 2021-07-07:
Sure I'll test it when the time comes

soostdijck, 2021-09-16:
> @soostdijck, can easily install by using the _from url..._ box in the plugin manager. Let me know if you're interested since I'll be looking for testers soon.

How it the release going? I'm still missing my thumbnails :(

scottmudge, 2021-09-17:
Does https://github.com/FormerLurker/ArcWelderPlugin/commit/103ad7d63de2c2cc76803ef7ad1e7a58abd79df1 also fix the issue with PrusaSlicer thumbnails being removed? I installed the latest RC of 1.1.0, but the thumbnails are still removed. 

If it's only compatible with Cura thumbnails, I'll go ahead and create a PR for a fix for PrusaSlicer thumbnails as well. Just want to make sure I'm not fixing something that's already fixed.

notfixingit3, 2021-09-19:
I am also having random issues with some thumbnails from PrusaSlicer being removed.

jneilliii, 2021-09-24:
> Does [103ad7d](https://github.com/FormerLurker/ArcWelderPlugin/commit/103ad7d63de2c2cc76803ef7ad1e7a58abd79df1) also fix the issue with PrusaSlicer thumbnails being removed? I installed the latest RC of 1.1.0, but the thumbnails are still removed.
> 
> If it's only compatible with Cura thumbnails, I'll go ahead and create a PR for a fix for PrusaSlicer thumbnails as well. Just want to make sure I'm not fixing something that's already fixed.

Based on what I'm seeing in that commit I think it is accounting for `Slicer Thumbnails` too.

ManuelMcLure, 2021-09-24:
With 1.1.0rc2 it _mostly_ works. However, sometimes (maybe 20% of the time) when AW automatically welds a file uploaded from SuperSlicer the thumbnail doesn't appear. Usually if I delete the welded file and use the button to re-weld the newly welded file will have the thumbnail as expected.

jneilliii, 2022-01-20:
are you able to rescan the files from the slicer thumbnails plugin's settings?

CBDesignS, 2022-01-20:
@jneilliii I have just started an 10hr print so do not want to upset the apple cart at the moment + I am fighting/testing a fix for a Marlin thermal protection bug that strikes randomly so as soon as the print completes or the bug wipes me out I will check. I also did just update to 1.7.3 to see if that may have fixed anything but its still the same.

CBDesignS, 2022-01-21:
Print completed and no thermal bug. yee. @jneilliii running a rescan in the slicer settings brung the missing thumbnail back. thx I will keep my eye open if they start going missing again. I ssh`d in and checked to see if the plugin had saved the png file and it did so dunno what is slightly wonky....

jneilliii, 2022-01-21:
could just be some race condition with the file list not getting updated. I have a request to re scan when pressing the file list refresh button I'm considering implementing this weekend. doesn't solve the problem, but could make it a little easier to rescan.