FormerLurker/ArcWelderPlugin #179: Unable to check for updates in OctoPrint

Taomyn, 2021-03-18:
Did not see any errors the last few days but this morning OctoPrint reports it cannot see update information for ArcWelder. All the other plugins show no error:

Please use the following:

```
2021-03-18 07:18:46,350 - octoprint.plugins.softwareupdate - ERROR - Could not check arc_welder for updates
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1496, in _get_current_version
    target, check, online=online, credentials=credentials
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/version_checks/github_release.py", line 305, in get_latest
    target
octoprint.plugins.softwareupdate.exceptions.ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None
```

seantapscott, 2021-03-18:
Hi Taomyn,

You can read through this issue: https://github.com/FormerLurker/ArcWelderPlugin/issues/165

Issue was reported as resolved already.  Need to install the development version of arcwelder as indicated here:
https://github.com/FormerLurker/ArcWelderPlugin/discussions/141

Or view the README file for more information on selecting the development release channel.

Taomyn, 2021-03-19:
@seantapscott thanks, updated and all looks ok now. Will report back if I have any issues with RC2.

closed 2021-03-19 by Taomyn