FormerLurker/ArcWelderPlugin #180: Installation fails - compile error

graffen, 2021-03-22:
I'm trying to install Arc Welder on a fresh (from last week) OctoPrint install. The install was done from the RPi image. I get the following log output when I try to install Arc Welder: 

```
2021-03-22 10:25:59,077   /home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpi0oeaz6g/ArcWelderPlugin-master.zip --no-cache-dir
2021-03-22 10:26:02,108 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
2021-03-22 10:26:02,109 > Processing /tmp/tmpi0oeaz6g/ArcWelderPlugin-master.zip
2021-03-22 10:26:06,633 > Requirement already satisfied: six in /home/pi/oprint/lib/python3.7/site-packages (from Arc-Welder==1.0.0+u.bb71e8f) (1.15.0)
2021-03-22 10:26:06,634 > Requirement already satisfied: OctoPrint>1.3.8 in /home/pi/oprint/lib/python3.7/site-packages (from Arc-Welder==1.0.0+u.bb71e8f) (1.5.3)
2021-03-22 10:26:06,636 > Requirement already satisfied: setuptools>=6.0 in /home/pi/oprint/lib/python3.7/site-packages (from Arc-Welder==1.0.0+u.bb71e8f) (51.1.2)
2021-03-22 10:26:06,636 > Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.12.1)
2021-03-22 10:26:06,636 > Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.18.2)
2021-03-22 10:26:06,637 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2021.2.4)
2021-03-22 10:26:06,637 > Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.6.0)
2021-03-22 10:26:06,637 > Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.11.2)
2021-03-22 10:26:06,637 > Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.25.1)
2021-03-22 10:26:06,637 > Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (3.5)
2021-03-22 10:26:06,638 > Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.19.5)
2021-03-22 10:26:06,638 > Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.2.0)
2021-03-22 10:26:06,638 > Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.4)
2021-03-22 10:26:06,639 > Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2021.2.23)
2021-03-22 10:26:06,639 > Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.1.5.post0)
2021-03-22 10:26:06,639 > Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.0.0)
2021-03-22 10:26:06,639 > Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.4.21)
2021-03-22 10:26:06,640 > Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.1)
2021-03-22 10:26:06,640 > Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.0)
2021-03-22 10:26:06,640 > Requirement already satisfied: watchdog==0.10.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.10.4)
2021-03-22 10:26:06,640 > Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.0.1)
2021-03-22 10:26:06,641 > Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.5.0)
2021-03-22 10:26:06,641 > Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.10.9)
2021-03-22 10:26:06,641 > Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (7.1.2)
2021-03-22 10:26:06,641 > Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.0)
2021-03-22 10:26:06,642 > Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.6.1)
2021-03-22 10:26:06,642 > Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2.8.5)
2021-03-22 10:26:06,642 > Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (3.1.1)
2021-03-22 10:26:06,642 > Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (5.1.1)
2021-03-22 10:26:06,643 > Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.8.0)
2021-03-22 10:26:06,643 > Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.1.2)
2021-03-22 10:26:06,643 > Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.2)
2021-03-22 10:26:06,643 > Requirement already satisfied: zeroconf<0.25,>=0.24 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.24.5)
2021-03-22 10:26:06,644 > Requirement already satisfied: feedparser<7,>=6.0.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (6.0.2)
2021-03-22 10:26:06,644 > Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (5.3.1)
2021-03-22 10:26:06,644 > Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (5.8.0)
2021-03-22 10:26:06,644 > Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (2020.11.13)
2021-03-22 10:26:06,645 > Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (4.0)
2021-03-22 10:26:06,645 > Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.57.0)
2021-03-22 10:26:06,645 > Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (1.0.7)
2021-03-22 10:26:06,645 > Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.1.1)
2021-03-22 10:26:06,646 > Requirement already satisfied: pyasn1>=0.1.3 in /home/pi/oprint/lib/python3.7/site-packages (from rsa==4.0->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.4.8)
2021-03-22 10:26:06,646 > Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from watchdog==0.10.4->OctoPrint>1.3.8->Arc-Welder==1.0.0+u.bb71e8f) (0.1.2)
[... cut ...] ild-z5v279mq/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-d8skgyd9/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python3.7/Arc-Welder Check the logs for full command output.
```
Am I missing something here? Google has not helped me an awful lot, so I decided to try here.

FormerLurker, 2021-03-30:
This is very strange.  There was an internal compiler error early on.  Will need some time to look at this, but you might try updating your OS and packages.

graffen, 2021-04-02:
I think everything is as up-to-date as it can be. I'm using the latest RPI image of Octoprint and have since run `apt update && apt upgrade`` to make sure everything is dandy. Anything else you'd like me to try? I'm glad to help figure this one out.

graffen, 2021-04-04:
Incidentally, I just noticed I see the same error when trying to install OctoLapse.