FormerLurker/ArcWelderPlugin #182: ArcWelder not producing printable file

JonathanPartain, 2021-03-28:
Hi! I recently installed ArcWelder on Octoprint, and have not been able to get gcode that looks good to print at all. First of all, I do not believe this is a firmware issue, as I have ARC enabled:
```
Send: M115
Recv: FIRMWARE_NAME:Marlin 2.0.7.3 Bugfix M.A.R.C. Manual Mesh, based upon current bugfix-2.0.x (Mar 29 2021 00:33:58) SOURCE_CODE_URL:github.com/mriscoc/Marlin_Ender3v2/tree/bugfix-2.0.x-ManualMesh PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 V2
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:1
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:0
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:1
Recv: Cap:SDCARD:1
Recv: Cap:REPEAT:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:0
```
As seen above, ARCS is enabled. 
I have also attached an image below, showing the diff between the ArcWelded file (left) and base gcode (right). 
All other settings are default.

Sliced with Cura version 4.7.1
[image]

FormerLurker, 2021-03-28:
Beware of gcode visualizers.  Most can't render arc commands.  If you post your gcode (use gist.github.com and drop a link), and I will render it for you to see if there is a problem.  Fyi, simplify 3d is the only slicer I know of that correctly renders g2/g3 commands as of now.

JonathanPartain, 2021-03-28:
@FormerLurker Thanks for the quick response! Here is a link to the gist: https://gist.github.com/JonathanPartain/8b374482861d111c0597aea0443d4f7b

seantapscott, 2021-03-29:
Hey Jonathan,

Local peer support dude here. Did you turn on the RX_BUFFER_SIZE setting?
I recently was troubleshooting an issue with the longer arc commands
creating buffer overflow issues, specifically on Creality boards v4.2.2 .

JonathanPartain, 2021-03-29:
@seantapscott Thanks for checking! I had not enabled that setting, so I probably should and then re-flash. The good news is that I tried printing the "broken" gcode, and it worked brilliantly, so it looks like it was just an angry cura. 

@FormerLurker Seeing as this was just a visual issue in cura, I will be closing this issue. Thanks for your help!

closed 2021-03-29 by JonathanPartain

FormerLurker, 2021-03-30:
You  beat me to it @JonathanPartain!  Glad your issue was just one of visualization.  I hear PrusaSlicer is going to be adding arc support for visualization, and hopefully cura will follow soon!

JonathanPartain, 2021-03-30:
@FormerLurker What I also try to do before printing is use the GCode viewer on octoprint as a sanity check. You can't visualize it as an entire model, but it works well enough to see that it's not doing anything crazy.

FormerLurker, 2021-03-30:
@JonathanPartain, you can view the entire model layer by layer.  Install the pretty gcode plugin to view the whole thing.  It supports arcs.