FormerLurker/ArcWelderPlugin #184: Surface became rugged

alwayslearningperson, 2021-03-30:
Hi, after using the plugin, the relatively smooth surface became rugged. Would you please check what is wrong? Thank you
[https://imgur.com/CS4SmGp](url)
[Dome_60_1.2mm_max_test15_0.2mm_PETG_MK3S_2h1m.gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/6232395/Dome_60_1.2mm_max_test15_0.2mm_PETG_MK3S_2h1m.gcode.zip)

FormerLurker, 2021-03-30:
The gcode looks fine after welding, so whatever the problem is, it is not obvious:

[image]

seantapscott, 2021-03-31:
What is your printer model and firmware version?

alwayslearningperson, 2021-03-31:
> What is your printer model and firmware version?
> […](#)

Prusa i3MK3S+. Firmware version is 3.9.3 which is the latest one.

alwayslearningperson, 2021-03-31:
> The gcode looks fine after welding, so whatever the problem is, it is not obvious:
> 
> [image]

Thanks. Why the top part does not look smooth, round and uniform overall? I cut a sphere into half to make it.

alwayslearningperson, 2021-03-31:
The problem is not caused by Arc Welder. Sorry