FormerLurker/ArcWelderPlugin #185: Arc Welder setting 'G90/G91 Influences Extruder'

b-morgan, 2021-03-31:
I get the following error message with Arc Welder v1.1.0rc2:

> Your firmware requires the 'G90/G91 Influences Extruder' setting to be ENABLED, but it is disabled. Edit the Arc Welder settings, enable the G90/G91 Influences Extruder setting, and run the firmware check again.

I can't find this setting in Arc Welder but the documentation on [https://awesomeopensource.com/project/FormerLurker/ArcWelderPlugin](url):

> Arc Welder needs to know one property of your printer's firmware to guarantee accurate results in all slicers and with all start/end GCode: G90/G91 influences extruder.

> - Use Octoprint Settings - Octoprint has a setting for G90/G91 influences extruder in the Features tab. Enabling Use Octoprint Printer Settings will cause Arc Welder to use OctoPrint's setting. Default: Enabled

> - G90/G91 Influence Extruder - If Use Octoprint Feature Settings is unchecked, Arc Welder will use this setting to determine if the G90/G91 command influences your extruder's axis mode. In general, Marlin 2.0 and forks should have this box checked. Many forks of Marlin 1.x should have this unchecked, like the Prusa MK2 and MK3. I will try to add a list of printers and the proper value for this setting at some point, as well as a GCode test script you can use to determine what setting to use. Keep in mind that most slicers produce code that will work fine no matter what setting you choose here. Default: Disabled

I believe the error message should be changed (conditioned) depending on the setting of `Use OctoPrint Settings`

`browser.user_agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:87.0) Gecko/20100101 Firefox/87.0
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1200
env.hardware.ram: 915058688
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 3 Model B Rev 1.2
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.2.4
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.0rc1
printer.firmware: Marlin
systeminfo.generator: systemapi`

FormerLurker, 2021-03-31:
Ok, I'll try making the error message more clear, thanks!

FormerLurker, 2021-03-31:
Actually, maybe instead of hiding g90/G91 influences extruder, I can show it under the 'Use Octoprint Settings' check box, but gray it out so that it is obviously dependent on 'Use OctoPrint Settings'?  Maybe I could do both even...

b-morgan, 2021-03-31:
I like the idea of showing it but grayed out but if you do show it under the 'Use OctoPrint Settings', it should reflect the state of the OctoPrint setting.

FormerLurker, 2021-03-31:
> 'Use OctoPrint Settings', it should reflect the state of the OctoPrint setting.

So maybe I can just add some verbiage, like ```Use Octoprint Settings - G91 Influences Extruder = True``` or something like that?

b-morgan, 2021-03-31:
If I uncheck 'Use OctoPrint Settings' then there are two checkboxes, one for 'Use OctoPrint Settings' and one for 'G90/G91 Influence Extruder'. I think the following will work.

- Instead of hiding the 'G90/G91...', gray it out but set the checkbox to checked if the Feature is checked and empty if the Feature is unchecked (i.e. the gray checkbox mirrors the OctoPrint setting).
- When the 'Use OctoPrint Settings' checkbox is unchecked, then ungray the 'G90/G91...' and the checkbox reflects the Arc Welder state.

For the error message, 'Edit the Arc Welder setting, ...' changes to 'Edit the Octoprint Feature setting, ...' (note the change from settings to setting) depending on which setting is actually being used.

I think this should reduce the confusion while putting the information in one place.

FormerLurker, 2021-03-31:
> When the 'Use OctoPrint Settings' checkbox is unchecked, then ungray the 'G90/G91...' and the checkbox reflects the Arc Welder state.

That's perfect, will implement, thanks!

FormerLurker, 2021-07-07:
I finally got around to implementing this.  I'm working on plugin issues and features now that some major core algorithm fixes and enhancements are complete.  Thanks for the suggestion!

FormerLurker, 2021-07-07:
I spoke too soon, forgot about the error notification message.  Will post back when that's fixed.

FormerLurker, 2021-07-07:
OK, I think that is fixed too.  Will do some testing over the next day or two.  The changes are on the devel branch.