FormerLurker/ArcWelderPlugin #192: "mm_per_arc_segment" setting not present

Eliminateur, 2021-04-18:
In the octoprint plugin there's no way to configure mm_per_arc_segment

closed 2021-04-18 by Eliminateur

FormerLurker, 2021-04-19:
I see this is closed, but I did add something I'm calling 'Firmware Compensation' that is available in the beta for people running Marlin 1 or a fork.  See the discussions tab above for instructions on how to install, and be sure to read the instructions within the settings.

Anyway, good luck!

Eliminateur, 2021-04-19:
i closed it because i realized that setting is ONLY if you need firmware compensation(documentation is kind of ambiguous about it).

For example, if in my firmware i change the mm per arc segment to 0.1, ¿do i need to set that in arc welder?

FormerLurker, 2021-04-19:
Here is the context help for the firmware compensation enabled setting:

[image]

Let me know how I can improve that to make it more clear.

> For example, if in my firmware i change the mm per arc segment to 0.1, ¿do i need to set that in arc welder?

You only need firmware compensation if your firmware ONLY has the mm_per_arc_segment setting (and not, say, minimum_arc_segments, or some of the other enhanced arc generation settings) AND you have issues where arcs of a small radius are coming out flat.  0.1mm segments is very small, so you probably won't notice the flattening at all.

Eliminateur, 2021-04-19:
perfect, so for marlin 2.0.2 that has both MM_PER_ARC_SEGMENT and MIN_ARC_SEGMENTS defined and configured in the config files then i should not enable anything on arc welder

FormerLurker, 2021-04-19:
That is correct!

Eliminateur, 2021-04-19:
Perfect,
that would be a good addendum to the documentation on fw compensation:
If you're running Marlin 2.0.x and have these two values set, then you don't need to set FW compensation or configure those two values(cura plugin has no fw compensation checkbox, only those values to edit or reset)