FormerLurker/ArcWelderPlugin #193: no big improve with cylinders

TylonHH, 2021-04-19:
Hey,
I'm using pre-release 1.1.0rc2 but I only get a reduction of 21% with an cylinder. I set the resolution to 0,1 and the tolerance to 5%. This is the maximum. I also see the path on the surface. It nearly should be a whole circle, shouldn't it?
Here is an example cylinder. _Print_ it with one/two perimeter.
[LID_vase_mode_deeper_notch.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/6333954/LID_vase_mode_deeper_notch.zip)

Best regards

TylonHH, 2021-04-19:
Here is an example printed in vase mode how it should look like. Here it decreased by nearly 95%
[image]

FormerLurker, 2021-04-19:
>  I set the resolution to 0,1 and the tolerance to 5%. This is the maximum.

But that is not the maximum:

[image]

This is an issue with the number of facets in your .stl or the resolution settings (or maximum deviation) in your slicer.  Just increase the resolution if you need to.  Let me know if that helps.

FormerLurker, 2021-07-06:
Are you still having issues, or is this issue closed?  Just checking in :)