FormerLurker/ArcWelderPlugin #194: Tank you, and kudos !

yet-another-average-joe, 2021-04-25:
I just discovered your new project, and installed your Cura plugin.
I just exported a few STLs with ultra high resolution options (0.02mm deviation, 1° resolution), activated Arc Welder in Cura, and got the best and smoothest prints I ever got, by far ! Thank you.
And kudos, because I know what you had to do : decades ago, I had to do something very similar, using 3rd order splines. (was part of my thesis work). I know the amount of work such a piece of software does imply. And I didn't have to parse GCode... Absolute coords came from a digitizer (just had to code a multi window GUI and a digitizer driver with Borland Turbo C). Didn't have to mess around with different languages, interfaces and APIs.

Thank you and kudos again.

You're a math head. And this is good.

Of course, you'll have to close this (non) issue as soon as you read it. Sorry, but until now, I didn't find any issue.
Just wanted you to know that some guys do know what game level you're playing.

yet-another-average-joe, 2021-04-30:
Time to close this (non) issue !

closed 2021-04-30 by yet-another-average-joe

FormerLurker, 2021-04-30:
Lol, I'm thought i responded to this!  Thank you for your compliments!

Also, it is funny, but my senior thesis used cubic splines as a function fitter for an AI to play a simplified version of Othello ;) arcwelder doesn't yet use this method, but I am working to add bezier curve support for the fun of it.

yet-another-average-joe, 2021-05-01:
Cubic splines are the minimal order needed to get tangents in any case (AFAIR). 2nd order does not work.

The book my work was based on : https://www.amazon.fr/Lutilisation-courbes-surfaces-en-CAO/dp/2866011392 (by Pierre Bézier). For... my medical thesis. Now retired. It was about describing the face contours, relative to the underlaying skull, about facial surgery (2D only, 30+ years ago, now retired).

I'm sure you will succeed with beziers. I did it after a couple years of maths and 10 years of medical studies. Will be easy for you if you are in maths.

Forgot to say : I didn't expect to run my old (heavily modified) Tevo Tornado so fast (up to 100 mm/s) with no polygons, with no stuttering, with very little ringing, and with so smooth results. Incredible. The limit is the frame and bed rigidity (at some point the nozzle hits the solidified plastic too hard due to vibrations)).
Why did I purchase a SKR 1.4 Pro ? (by the way still in its box ; not really  : hooked to a breadboard ;) ).

OctoLapse, ArcWelder, what's next ?

An idea : 

I'm working from time to time on a Marlin UI emulator for OctoPrint ; but I'm unable to code a plugin (could just set some color options in a page derived from hello example) ; I'm using a STM32 to turn the Pi into a SPI slave, and currently displaying Marlin as an overlay using DispManX ; previous version using the frame buffer can be seen here : https://www.youtube.com/watch?v=vd2Wgo4UeBw ; if you are out of projects... The hardware and the low level software (STM32 and RasPi) are working... Don't know how to display this in a OctoPrint tab, and had no success on the OctoPrint forums...

(sorry for the flooding)