FormerLurker/ArcWelderPlugin #196: Major Slicing error

Doyle884, 2021-05-10:
Every time I slice with arc welder enabled I end up with a jumbled mess of random lines and missing sections in preview and looks like file corruption. When ever I print, it prints the jumbled mess. When ever I turn it off, slices just fine

FormerLurker, 2021-05-10:
Need more info.  Without that, my best guess is that arcs are disabled in your firmware (99% of the time, this is the issue).  Send the following command to your printer:

```G2```

If your printer responds with ```Unknown Command``` and/or disconnects, then arcs are disabled.  If it responds with ```Bad Parameters``` arcs are enabled and you can supply more information.  At a minimum I need your source and arc-welded gcode file.

Doyle884, 2021-05-14:
It's before it even gets to the printer, I have also used it before without issue and it was working great by the way!, I haven't gone hardwired or octopi yet, I was using Cura to slice and I would save it to either my sd card or local disk and even viewing the Gcode in preview right after it's sliced it's all "messed up". I will send pictures of gcode preview in cura and slic3r and the 
printed benchy in the morning.(Est)

FormerLurker, 2021-05-14:
Cura and slic3r cant properly display g2/g3 commands.  Please post the gcode to gist.github.com and i will run it through a visualizer that works.