FormerLurker/ArcWelderPlugin #199: When uploading multiple files to octoprint, the completed AW stats page is not visible

jurassic73, 2021-06-08:
I uploaded two files (drag and drop in one action) and both items were processed by AW however only the last one processed displayed the 'page icon' in the directory to then be able to see it's stats.  The other processed file with aw in the filename still had the two arrows icon.

Excellent plug-in BTW!  Amazing work.

FormerLurker, 2021-06-08:
Are you using the stable version or the latest development version?  A snapshot of the ArcWelder tab should suffice (it has the version number at the bottom including a link)

jurassic73, 2021-06-08:
Stable as pulled in via Octoprint I believe.  Arc Welder v1.0.0+u.bb71e8f

[image]

FormerLurker, 2021-06-08:
Yeah, that's a but I believe I fixed.  Try installing the latest beta version [using this guide](https://github.com/FormerLurker/ArcWelderPlugin/discussions/141).  FYI, I'm going to be pushing another major update soon with a bunch of fixes and enhancements.  Also, I recommend setting the max arc radius to 10000mm, as a few issues were discovered with marlin's interpolation algorithm, and that will work around them for now.  Newer versions will apply this automatically.

jurassic73, 2021-06-08:
Thanks for looking at this.  Appreciate it and will do.

closed 2021-06-08 by jurassic73

FormerLurker, 2021-06-08:
Of course!  If the new version doesn't work, reopen the issue and I'll double check the latest dev code.