FormerLurker/ArcWelderPlugin #204: Found a case where ArcWelder seems to output invalid results.

wirehead, 2021-07-07:
I was trying to print https://www.thingiverse.com/thing:4876137 on my new Ender 3 v2.  Because, of course, what point is having a 3D printer if you don't print at least five mods and eight test prints and never actually get around to printing anything useful?

After a few tries at printing it where it did a lot of stringy weirdness, the thought occurred to me that maybe I found a nice edge case in ArcWelder so, just in case that was actually what was going on, I tried printing the non-ArcWelder version.

But.. this is weird.  I've been running ArcWelder pretty much exclusively and all of my other prints aren't doing this.  Something about this particular piece seems to be messing it up.

I am using the 1.0.2 version of the Ender 3v2 firmware which should be a Marlin 2.x.  I'm using SuperSlicer 2.3.56.  I'm using OctoPrint 1.6.1 with Python 3.7.3 and Arc Welder v1.0.0+u.bb71e8f.

I've uploaded the gcode file that ArcWelder seems to be acting up on and a photo of the failed print with a successful un-arc-welded print next to it:

[image]
[Filament_Guide_v3_0.25mm_PLA_ENDER3V2_57m.gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/6774431/Filament_Guide_v3_0.25mm_PLA_ENDER3V2_57m.gcode.zip)

FormerLurker, 2021-07-07:
Can you also attach the welded file please?

seantapscott, 2021-07-07:
Looks like buffer overflow.  Probably should switch away from Brims and
towards using an adhesive like Magigoo, as well, but that's not relevant to
the issue.

To fix buffer overflow, you'd have to compile from Marlin bugfix branch.

FormerLurker, 2021-07-07:
Could also be an arc going in the wrong direction.  This is rare in this version of arc welder, and I can tell by looking at the welded gcode file.  It is worth eliminating that possibility since that has been fixed.