FormerLurker/ArcWelderPlugin #205: Abandoned?

holocronology, 2021-07-25:
Has this plug-in been abandoned? It's been a while since there has been any new release or issues closed.

ldursw, 2021-07-25:
The development is still active in the [devel](https://github.com/FormerLurker/ArcWelderPlugin/tree/devel) branch. There are a lot of new features since the latest stable release but it may contain some bugs. You can help by installing the latest version and reporting any issue you find.

If you want to try it out you can install from the plugin manager by clicking 'Get More' then adding this line to the 'From Url...' box and clicking install: https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/heads/devel.zip

Once the install is finished, restart OctoPrint.

holocronology, 2021-07-25:
Cool. Thanks.

closed 2021-07-25 by holocronology