FormerLurker/ArcWelderPlugin #206: Going to push 1.1 to release?

eblieb, 2021-07-28:
Was wondering if you were going to push 1.1.0 to release.

FormerLurker, 2021-07-28:
Hopefully soon.  I am trying to test the updates for compatibility with Cura Arachne now, and also the conversion of travel moves.  I have discovered that some small adjustments might need to be made for Arachne (variable line width), though the changes are just minor tweaks.  I'll try to post the new version containing these changes tomorrow.

It's been way too long since I've pushed a live update, but there were a lot of strange issues I have been dealing with.  Turns out the vast majority are firmware problems, some of which have been corrected via firmware updates, and others required workarounds or settings changes to correct.

Thanks for your post!  If you feel like helping with the testing please let me know!

seantapscott, 2021-07-28:
All the firmware issues that I've seen are where the users are told to
compile a certain branch, they don't, and then they complain that the
firmware fix didn't work.

I also wonder if you should even be worrying about arachne.  Though it is
interesting, it is probably going to be in development for awhile, so is
largely an experimental release.

Just sharing grains of salt.

FormerLurker, 2021-07-28:
> I also wonder if you should even be worrying about arachne.  Though it is
interesting, it is probably going to be in development for awhile, so is
largely an experimental release.

There was a ton of hype about it when I started adding support, but perhaps I jumped the gun.  However, there are other benefits of adding flow rate change detection.  While developing support for this feature I found inconsistent flowrates (mm extruded/distance traveled) even in older versions of cura and slic3r.  They are sometimes massive (more than 10-20%).  Merging these commands into arcs results in the average flowrate being used, and for very small arcs that could affect quality.  I'm not sure what causes these variations exactly.

If I'm being honest though, I have just been nervous about updating this app since the original welding algorithm seems to work pretty well, but I've decided to get over it.  I'll post an RC ASAP and give it a week before pushing, provided there are no catastrophic issues.  I'll just push the console version to master without any RC since it is a lot simpler.

The [firmware library](https://github.com/FormerLurker/ArcWelderPluginFirmwareInfo) probably needs some updates.  Fortunately, that can be done outside of Arc Welder since it downloads the latest files automatically.   That means there is no hold up there.

Thanks for your input @seantapscott, as always.

eblieb, 2021-07-28:
Ya I have been having a hell of a time getting my Ender 3 V2 to have a proper flow rate and quality has never been exactly what I think it should be. Wasn't sure if it was arcwelder doing it or what. I just removed my all metal hotend and went back to the stock hotend with a quality PTFE tube but still having massive stringing issues.

CRCinAU, 2021-08-15:
I was just thinking this myself - as if you recall, the original 1.0.0 release was broken for Octoprint update manager....

1.1.0-rc1/rc2 fixed this, but when you install directly via Octoprints plugin directory, you get the broken update version.

So there'd be many benefits from pushing the update...