FormerLurker/ArcWelderPlugin #207: Wrong prints with ArcWelder since couple of days ago

elpeter25, 2021-08-01:
Hi,

I've been using ArWelder for very long time and was so happy with the results it gave. But since a couple of days without changing anything in the printer nor in the slicer arcs started to look awful. See picture below:

[image]

The one on the left is with ArcWelder 5 days ago, the one on the right is with ArcWelder yesterday.

What can possibly be happening? 

Not sure if relevant, but this is latest update from system:

Start-Date: 2021-07-25  23:04:44
Commandline: apt upgrade
Requested-By: peter (1000)
Upgrade: initramfs-tools-bin:amd64 (0.130ubuntu3.12, 0.130ubuntu3.13), gzip:amd64 (1.6-5ubuntu1, 1.6-5ubuntu1.1), initramfs-tools-core:amd64 (0.130ubuntu3.12, 0.130ubuntu3.13), initramfs-tools:amd64 (0.130ubuntu3.12, 0.130ubuntu3.13)
End-Date: 2021-07-25  23:05:16

I'm using Octorpint over Ubuntu and latest version of the plugin.

Cheers!

elpeter25, 2021-08-01:
More info. Here you go properly working arcwelder gcode on a Prusa MK3s from 6 days ago and also the ones not working on a Prusa MINI, (it's not the printer, I can upload the gcode again in mk3s to be processed by arcwelder but don't want to override the one working)
And following pic is from those MINI gcodes just printed couple of minutes ago.
[image]
[ArcWelder issue.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/6912598/ArcWelder.issue.zip)

FormerLurker, 2021-08-01:
> I'm using Octorpint over Ubuntu and latest version of the plugin.

What version of the arc welder plugin are you using?  I'm guessing this is the Octoprint Plugin (not the cura plugin or console version) we are talking about, correct (I have to double check)?  You can find the version of the octoprint plugin on the ArcWelder tab:

[image]

How we debug this depends on the version in question, but the mini needs the G90/G91 influences extruder setting to be true, and the MK3 needs this to be false.  Also, it's hard to tell the scale of these prints, but can you measure the flat edges on the mini version (the bits that should be curved, but are not)and see if they are approx 1 mm (between 0.5mm and 2mm)?  That will help as well. 

Without more info, it looks like a firmware issue to me, but there is a setting in the dev version of arcwelder to compensate for this issue.  You could try printing a larger curve (maybe a big circle) on the mini with welded gcode and see if that looks better.  That would be a definitive test.

Thanks for your issue!