FormerLurker/ArcWelderPlugin #210: Error in Software update tab

mcdowelljc, 2021-08-16:
**Describe the bug**
Arc Welder: Anti-Stutter: unknown 
Release Channel: Stable
 Installed: ?
Available: ?
Unknown error while checking for update, please check the logs

**Expected behavior**

**Debug Log**
2021-08-15 20:41:56,162 - octoprint.plugins.softwareupdate - ERROR - Could not check arc_welder for updates Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1785, in get_current_version
    target, check, online=online, credentials=credentials
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/version_checks/github_release.py", line 305, in get_latest
    target
octoprint.plugins.softwareupdate.exceptions.ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None

FormerLurker, 2021-08-16:
Yeah, I am aware of this, and it has been corrected, but I need to submit a pull request to Octoprint to fix the current master version unfortunately :(

mcdowelljc, 2021-08-16:
It's all good.   I appreciate all that you have done!   It has made my printing MUCH faster!

FormerLurker, 2021-08-16:
>  I appreciate all that you have done! It has made my printing MUCH faster!

I'm so happy it's helped you out!  Thank you for the compliment :)

Swap-File, 2021-12-08:
Is this error still expected behavior?  Just wanted to make sure I didn't have a problem in my OctoPi setup.

FormerLurker, 2021-12-08:
It is for v 1.0.  I still need to submit a PR to the OctoPrint repo, which will hopefully not take too much longer.

richard-scott, 2022-02-15:
It's strange that you can install from [here](https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/tags/1.1.0rc4.zip) and the Software Update section reports the correct values. 

[image]