FormerLurker/ArcWelderPlugin #213: Weird effect of AW on small circular paths

comatose-tortoise, 2021-08-25:
Had a lot of problems with a model I was printing, thought it was my printer settings that were off until I discovered that ArcWelder was totally messing up small details in the form of circles.

Prusa MK3S+
PETG
AW resolution: 0.1 mm
AW maximum arc radius: 1000000 km

This is the exact same gcode, only difference is that one is processed through AW (the octoprint plugin version):
[image]
 
Apart from the out-of-focus photo and the terrible stringing, the unaltered gcode actually produces round features in the print, while the AW-processed gcode produces what looks more like straight lines crossed over each other. 

If I look at the AW-processed gcode in the octoprint viewer (does anybody know of any other gcode viewer that can correctly display AW-processed code? I haven't found one), it looks like this, with the circles jumping around a bit:
[image]

The unaltered gcode has static circles when moving through the layers:
[image]

FormerLurker, 2021-08-25:
I will look into the offset circle, but I do know why the lines look straight instead of round.  It is firmware related (I have submitted a PR to fix this that will hopefully be merged in some day).  There is a workaround that I will post asap.

comatose-tortoise, 2021-08-25:
Glad to hear there is a fix! Looking forward to the workaround while waiting for the merge.

So firmware related, that means this is a problem with Prusa printer firmware, not affecting other brands?

FormerLurker, 2021-08-25:
> that means this is a problem with Prusa printer firmware, not affecting other brands

Well, I wouldn't go so far as to say it's a problem exactly, but rather a limitation of the arc interpolation algorithm.  It is configured to draw lines of 1mm (up to 1.999mm actually) , which is too long for circles of a small radius.  Marlin 2.0 added some other options to improve this for arcs of a smaller radius.  It affects any firmware running a fork of Marlin 1, and depending on the configuration it can affect Marlin 2 as well, though the defaults will prevent issues.

The workaround is something I am calling 'Firmware Compensation'.  You can set a value for the minimum number of arc segments that should be generated for a given circle, and ArcWelder will break up any arcs as necessary to prevent these flat segments from being produced.  It does lower the compression somewhat, but only for arcs with a small radius.  It's available in the devel branch, and I am hard at work on a release.  I'm testing some new features now, and if you want to wait for a bit, I will let you know when they are ready if you're interested.

FormerLurker, 2021-08-25:
OH, can you post the before and after gcode for the above images?  I'd like to see what's up with that offset circle.  It could be a glitch in arcwelder (might have even been fixed already, if that's the case), or it could be an issue with the visualizer.  I'd  like to test all possibilities.

comatose-tortoise, 2021-08-25:
Sounds good! How long until release you think?

Here are the files:

[210825_AW_small_circles_gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/7047589/210825_AW_small_circles_gcode.zip)

FormerLurker, 2021-08-25:
Prob a few weeks.  If you want, you can try [this version](https://github.com/FormerLurker/ArcWelderPlugin/releases/tag/1.1.0rc2), which has most of the features of the one I'm working on, including a firmware checker and firmware compensation.  It should warn you about the issue you are experiencing.

comatose-tortoise, 2021-08-25:
Cool, I'd like to try that, but I don't really know how to. Is it possible to install it through the octoprint UI?

FormerLurker, 2021-08-25:
Yeah, you just paste this link into the plugin manager's *From Url...* text box and click install:  https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/tags/1.1.0rc2.zip

comatose-tortoise, 2021-08-25:
Great, I'll try that tomorrow and see if there's an effect on this print!

FormerLurker, 2021-08-25:
Make sure you enable firmware compensation.  The defaults should work OK.  Msg me if you have any questions.

FormerLurker, 2021-08-30:
> BTW, what does it mean that the firmware converts the arcs back into segments? Won't that defeat the purpose?

The original purpose of ArcWelder was to reduce serial traffic when streaming a print via USB.  This can reduce stalling caused by planner starvation, which causes all kinds of quality issues.  However, there are other advantages depending on your firmware.  G2,G3 can be configured to print as many segments as possible, potentially increasing quality, but older implementations probably won't give you much of a boost.

In other words, getting better quality when planner starvation isn't involved is mostly firmware/board dependent.  I hope that makes sense.

comatose-tortoise, 2021-08-30:
Is there any printer firmware that can use arcs without converting them back to straight lines?

FormerLurker, 2021-08-30:
> Is there any printer firmware that can use arcs without converting them back to straight lines?

No, but then again you're only going to be printing an approximation of a shape no matter what you do, or what firmware you're running.  This is also true for G1 commands as well.  The best you can do is to approximate the shape you want to print with the maximum resolution your equipment can produce.

For example, many slicers will restrict the minimum amount of motion that can be made.  Cura's default is 0.5mm (there is also a max deviation setting, so it's a bit complicated).  It used to be 0.1mm, but they turned it down because many printers couldn't handle the detail without planner starvation, which causes lots of difficult to diagnose quality issues.  My firmware is set to print a maximum of 1mm segments and a minimum of 0.1mm segments, and a minimum of 48 segments in a full circle, which is pretty good.  If you've got a good board, you can probably configure it to have even better resolution than that.

Also, you might be interested to know that a gcode to adjust arc settings dynamically might finally be on the horizon.  That should let one test arc settings and get the best possible detail for curves without planner issues.

jbuszkie, 2021-09-03:
I just hit this.. 
[image]
Maybe I'll try the new version as well!  I was tearing my hair out with printer settings and a nozzle change and filament!! I would not have thought arc-welder was the culprit!!

comatose-tortoise, 2021-09-03:
That's exactly how it was for me too!

FormerLurker, 2021-09-03:
> I would not have thought arc-welder was the culprit!!

To be fair, this is actually a firmware issue.  I added the workaround (firmware compensation) to deal with it as best as I could.

jbuszkie, 2021-09-03:
Sorry, I didn't mean to imply any shortcomings of arc welder. It was late last night when I saw this and quickly read through this.
Is this a know firmware issue?  Has this been fixed in a newer Marlin firmware?  I have not tried to update my firmware in a long time!

FormerLurker, 2021-09-03:
That's ok, just wanted to clarify.

Also, it isn't exactly a bug, but rather a shortcoming of older firmware settings.  Newer versions of marlin have a define called min_arc_segments, that will crank up the resolution for smaller arcs.

comatose-tortoise, 2021-09-03:
Absolutely no complaints about arc welder, you have also been extremely helpful. Thanks!

jbuszkie, 2021-09-03:
It looks like I'm running Marlin 1.1.9  So I don't have any of the Marlin 2 features..
So this workaround will (hopefully) have to do!  Thanks for helping with this!

mcdowelljc, 2021-09-06:
> It looks like I'm running Marlin 1.1.9 So I don't have any of the Marlin 2 features..
> So this workaround will (hopefully) have to do! Thanks for helping with this!

You really should upgrade your firmware!  There are several things that have been changed and added in the firmware's!  
I have an Ender 3 Pro with the  BTT SKR mini e3 v2.0 motherboard.   I have to build the firmware myself based on that and using the BL Touch probe.  
Many of your prints may change to looking better and printed a little faster.

jbuszkie, 2021-09-07:
I'll have to investigate that.  I'm using someone else's custom compiled firmware for my sidewinder. I haven't had the need to upgrade so far.  I thought about playing with kilpper..  We'll see.