FormerLurker/ArcWelderPlugin #218: Instructions surrounding MM_PER_ARC_SEGMENT

tommylux, 2021-12-24:
About tab explains:
Marlin has an option called MM_PER_ARC_SEGMENT that defaults to 1mm
this option has now been replaced with MIN_ARC_SEGMENT_MM and MAX_ARC_SEGMENT_MM. What is the associated feedback to set these to?

FormerLurker, 2021-12-24:
If those options are available, it means you are running a very recent version of marlin 2, and likely don't need firmware compensation enabled.  However, if you do, just leave the # setting at 1 for now.  Eventually I will get around to making firmware compensation more flexible.

FormerLurker, 2022-05-19:
Sure!  I assume you are using prusa firmware?  If so, the defaults are fine, I tested them thoroughly.

If not, it depends on your hardware.  1mm max is fine.  A min of 0.1 is ideal, but many 8 bit boards will struggle with values lower than 5.

If you give me more info regarding your setup, i could give more specific advice.

skl111, 2022-05-19:
It does, and do you have a link or info on these and how to use P in cura?
//#define ARC_SEGMENTS_PER_SEC 50 // Use the feedrate to choose the segment length
//#define ARC_P_CIRCLES // Enable the 'P' parameter to specify complete circles

FormerLurker, 2022-05-19:
You never want full circles (p) when 3d printing, else you will get overlap, so it should be good as is.  Neither cura nor arcwelder will make full circles.

Also, fyi, do NOT enable firmware compensation in arcwelder.  Your firmware is recent, and compensation will reduce quality for newer firmware, like yours.