FormerLurker/ArcWelderPlugin #219: PermissionError: [WinError 32] The process cannot access the file because it is being used by another process

jk7gr, 2022-01-06:
Dear @FormerLurker ,
Great work here. I am utilizing the plugin on 2 octoprint instances I have in my environment. 
While on the Raspberry PI the operation of the plugin is flawless on my Windows 10 instance I am facing some strange behaviour.
When I configure the plugin to rename the uploaded file and remove the old one I see that after file processing the new file is loaded and starts to print but the old file remains. If I check the log I see the error: 
"PermissionError: [WinError 32] The process cannot access the file because it is being used by another process" followed by the path and the name of the file I uploaded. 
It seems the issue is related also with one Octoprint file as well cause I also receive the below message just before the error message:
"  File "c:\OctoPrint\venv\lib\site-packages\octoprint\filemanager\storage.py", line 960, in remove_file
    os.remove(file_path)"
The same behavior is happening even if I select to replace the file that is uploaded with the processed file. In this case the situation is even more problematic as the actual file that is going for printing is the uploaded file and not the processed.

FormerLurker, 2022-01-07:
Yes, this has been a huge source of irritation for me.  I've even added some advanced 'retry' capabilities to the file delete to work around it, but no matter what I do I can't seem to get it to work in all cases.

So, this is a known issue in windows, but I've not been able to come up with any solution yet.  I'll keep this around so it is formally in the issues, and will update when/if I can find a solution.

ScrimpyD, 2022-01-17:
Hi @FormerLurker ,
I am also seeing this error on a Windows 11 system. The conversion is successful, the file prints just fine.

From the verbose log I see this at the very end of the conversion process:
2022-01-16 23:14:55,741 - arc_welder.gcode_conversion - INFO - py_gcode_arc_converter.ConvertFile - Arc Conversion Complete.
2022-01-16 23:14:55,742 - arc_welder.__init__ - INFO - Arc compression complete, creating a new gcode file: Z_Axis_Motor_Mount_Right_-_Short.aw.gcode
2022-01-16 23:14:56,138 - arc_welder.preprocessor - ERROR - An unhandled exception occurred while preprocessing the gcode file.
Traceback (most recent call last):
  File "d:\OctoPrint\venv\lib\site-packages\octoprint_arc_welder\preprocessor.py", line 105, in run
    self._process(path, processor_args, additional_metadata, is_manual_request)
  File "d:\OctoPrint\venv\lib\site-packages\octoprint_arc_welder\preprocessor.py", line 175, in _process
    pathlib.Path.unlink(self._target_file_path)
PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'C:\\Users\\Me\\AppData\\Roaming\\OctoPrint\\data\\arc_welder\\target.gcode'

This is just an idea, but how about instead of deleting the target.gcode file immediately after the conversion you leave it. When you start the next file conversion you can first check if that file exists and delete it then continue with the conversion.

Another thing you can try is to have a small 'pause' right before you delete the file. I have run into a similar issue on a Windows system when using .NET when trying to remove a file from a network shared folder. I would up just adding a small pause, 2 seconds, and that took care of the problem for me. It seemed that for whatever the reason Windows was holding the file as 'in use' after I closed it and tried to delete it.

Cheers,
Dan