FormerLurker/ArcWelderPlugin #220: octoprint.plugins.softwareupdate.exceptions.ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None

bruvv, 2022-01-08:
```
2022-01-08 11:07:53,298 - octoprint.plugins.softwareupdate - ERROR - Could not check arc_welder for updates
Traceback (most recent call last):
  File "/data/python/octoprint/lib/python3.9/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1784, in _get_current_version
    information, is_current = version_checker.get_latest(
  File "/data/python/octoprint/lib/python3.9/site-packages/octoprint/plugins/softwareupdate/version_checks/github_release.py", line 303, in get_latest
    raise ConfigurationInvalid(
octoprint.plugins.softwareupdate.exceptions.ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None
```

FormerLurker, 2022-01-08:
Sorry guys and gals, this is a known issue that will require either a manual update, or a change to the OctoPrint source (which I am hoping to do).  I made a small mistake with big ramifications when I first deployed this plugin, missing one bit of info in the software update settings, and it can't be fixed for existing installs without a pull request for OctoPrint itself.  The change is minor (basically adding a default setting in case of a null value), but it will have to coincide with an OctoPrint release.

nutsofamily, 2022-01-08:
So to clarify is it possible to fix ourselves manually with a different link?

FormerLurker, 2022-01-08:
yeah, as long as you're willing to install a pre-release version.  The next release isn't 100% finished (mostly due to time constraints I have with work), but it's extremely close.  The conversion algorithm has been used in the ArcWelder cura plugin for a while now, and seems to be working great.  You can install the rc/maintenance version with the following link via the OctoPrint plugin manager (get more->from url...):

```
https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/heads/rc/maintenance.zip
```

If you try it out, I'd appreciate some feedback.  Please note that the firmware checker hasn't been updated with new firmware versions in a while, but I'll get to that :)

Also, you might want to check out the [releases](https://github.com/FormerLurker/ArcWelderPlugin/releases) to see what changes you can expect.  The major improvements are firmware checking (for compatibility with arcwelder and g2/g3 commands), Travel move encoding and 3D arc encoding (for vase mode), support for variable line width (cura arachne and others), a better GUI and many more options.

bruvv, 2022-01-09:
thanks i've installed it and will see if there is any weird stuff going on. Thanks for the good explanation !

Btw small off topic question: would you rather recommend the cura extension ?

FormerLurker, 2022-01-09:
> it is showing up as Unknown.

Yeah, that is standard when installing straight from a github code branch zip.

> Btw small off topic question: would you rather recommend the cura extension ?

Well, that's up to you.  You're welcome to try both.  There are pros and cons for each.  The main benefit of the OctoPrint version is that there are more features and better statistics so you know what's going on.  The advantage of the cura plugin is that it is generally more friendly with complicated workflows, and probably a bit faster since you're probably slicing on a more powerful machine that the one running OctoPrint (probably a pi, but not necessarily).

> but after restarting it is showing up correctly

That too is expected since I fixed the issue in my code that wasn't supplying all of the necessary information to the software update plugin, but it's always good to get confirmation!

I'd also recommend looking through the settings just to see what's there.  Let me know if it didn't carry over any of your settings properly.  I haven't had a lot of time to test that part of the upgrade.  Should work, but never hurts to check!

bruvv, 2022-01-15:
Thanks for explaining, updated everything and the issue seems to be gone :) thanks

closed 2022-01-15 by bruvv