FormerLurker/ArcWelderPlugin #222: Bpos x square Collision

powasaurus-rex, 2022-01-31:
The past couple prints have failed with a "bpos x square" (and one time the _pos y square) error causing the print head to crash into the frame. The issue doesn't arise when using the original (non-arcweld) gcode file. 

I have been using AW on my delta printer for a year or so, this is the first time these errors have come up. All of the failed prints have been multiple pieces printed one at a time, though always fails ~%50 through the first piece.

powasaurus-rex, 2022-01-31:
I believe these are the correct files, I've uploaded and reuploaded so many that I can't 100% confirm without trying to print again. Currently using Arc-Welder (1.0.0+u.bb71e8f). Let me know what else you need.

[d12 v5.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/7973960/d12.v5.txt)
[d12_v5.aw.txt](https://github.com/FormerLurker/ArcWelderPlugin/files/7973961/d12_v5.aw.txt)

FormerLurker, 2022-01-31:
Awesome!  Just FYI, it's likely this has been fixed already on an as of yet unreleased maintenance build.  If you have the time, inclination, and patience to try it out, you can install it from the plugin manager by pasting in the following into the 'from url...' box:

```https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/heads/rc/maintenance.zip```

It may take me a bit to get around to this due to crazy things happening in my life right now, but it's on my list.  The next arcwelder and octolapse release is next on my todo list, and honestly I can't wait to get back to working on it again!

powasaurus-rex, 2022-01-31:
Cool, I'll take a look at that later this week. Totally get that life is busy because same.

ElDavoo, 2022-06-12:
> Cool, I'll take a look at that later this week. Totally get that life is busy because same.

So, did you take a look?