FormerLurker/ArcWelderPlugin #228: Layer shift

madmic1314, 2022-04-18:
Have found the strange layer shift issue (https://github.com/FormerLurker/ArcWelderPlugin/issues/113) when models have a straight edge - most noticeable with shelf brackets.

Using Cura plug-in and also Octoprint plug-in - both seem to do the same. If I process with the octo plug-in then download the gcode you can see the layer shift present. Seems to be along the layers which have the screw holes in.

seantapscott, 2022-04-18:
Couple of information gathering questions.

What's your printer model?  What firmware version is it?

Do you have dynamic precision turned on for Arc Welder?

Hypothesis
I'll bet those straight edges have longer length arc commands that are
causing truncation of the commands when sent to the printer planner.  An
update to the firmware and an increase to possible command length is needed
for best arc compatibility. But this is just an hypothesis.

madmic1314, 2022-04-19:
Thanks - 
Printer is BTT SKR 1.4 running marlin 2.0.9.3.

[100mm Bracket.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/8511622/100mm.Bracket.zip)

Attached is an example of AW in Octo - the arcswelded file as .aw in the name. You can see where the screw holes are there is a difference (know Cura can't show arc gcode yet).