FormerLurker/ArcWelderPlugin #229: Really bad layer shift on ender 3 pro marlin

Zchulski, 2022-04-26:
[image]

37 hours into this print with arc welder enabled and this happened.  Arc support in marlin is enabled I checked double checked and triple checked I don't know of any other reason this would be happening.  This ruined a 40 hour print I'm pissed

FormerLurker, 2022-04-26:
Just based on the photo, it looks like a layer shift.  Anyway, post the gcode if you can, and I'll run it through a visualizer to try to find differences.

Zchulski, 2022-04-26:
here is the gcode that i ran
[CE3PRO_Fan_With_Supports.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/8564456/CE3PRO_Fan_With_Supports.zip)

FormerLurker, 2022-04-26:
I don't see any obvious problems with the gcode toolpaths:

[image]

If there was a layer shift caused by arcwelder, I would expect to see odd toolpaths that would bring the printhead out-of-bounds, but I'm not seeing that here at all.

However, [take a look at this issue](https://github.com/FormerLurker/ArcWelderPlugin/issues/189).  It appears as though some firmware is misconfigured so that valid, but long, gcode commands can cause internal firmware corruption.  There are several ways to mitegate this, and some new ones in the development build:

1.  Set the 'Maximum Arc Radius' to 999.  This will shorten the maximum number of characters in the I and J components.
2.  Check your firmware configuration and fix the issues (see that link).
3.  Update arc welder to the development version (let me know if you're interested), where I've added a 'maximum gcode length' setting.  I'd recommend changing the maximum arc radius too (item  1).

I found gcodes longer than 62 characters in your file, so it's possible.  you could send some g1 commands to the printer, adding many decimals (i.e. G1 X0.000000000000000000000000000000000000000 Y0.000000000000000000000000000000000000000) and see what happens.

Zchulski, 2022-04-26:
I would not be against the dev build please send

FormerLurker, 2022-04-26:
Install from the plugin manager, and paste this into the 'From Url...' box after clicking 'Get More':

https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/heads/rc/maintenance.zip

Also, FYI, I had a strange issue where the install showed that it failed, but it actually worked.  Hoping it was a fluke.  Check the settings and verify that you have these options:

[image]

Also, the plugin version isn't correct (still a WIP).  That will be resolved soon, but fortunately it doesn't negatively affect anything.

FormerLurker, 2022-04-26:
Oh, and I should mention, you will need to re-convert that file to try again.  Send it to me first, and I'll take a look at the result, figure out the maximum line length, and hopefully we can avoid another failure.  On a print this long, we shouldn't take any chances.

Zchulski, 2022-04-26:
[CE3PRO_Fan_With_Supports.aw.gcode.zip](https://github.com/FormerLurker/ArcWelderPlugin/files/8564977/CE3PRO_Fan_With_Supports.aw.gcode.zip)

Zchulski, 2022-04-26:
> The toolpaths look fine, as in the last slice. Here is the longest gcode line:
>
> G3 X132.503 Y126.642 I-766.510 J-204.306 E1838.63786
>
> It is much shorter, but here's (minus the incorrect checksum and inexact line number, which I fudged into the longest possible value) the gocde sent from OctoPrint:
>
> N1378277 G3 X132.503 Y126.642 I-766.510 J-204.306 E1838.63786*255
>
> The 'shortest' line that has ever triggered this issue (found using TH3D firmware, are you using this by chance?) was 64 characters long. The above gcode is 66 characters. I would advise reducing the maximum gcode length setting to 50, which would have prevented this arc from being generated. This will slightly reduce compression, but not too much (arcwelder will show you how many arcs were aborted).
>
> Send me the updated file, and i'll keep repeating this process until the longest possible gcode sent from Octoprint is 64 characters. OR, you can print from SD, which omits the line number and checksum values. Let me know, and thanks!

I am going to print from an SD as I can't even make a connection to my printer with octo anymore for some reason

Sent from ProtonMail mobile

-------- Original Message --------

Zchulski, 2022-04-26:
> Good luck, and sorry about the previous failure!

Shit happens I guess lol. As for my firmware I'm not sore what I'm using. The latest marlin dev build

Sent from ProtonMail mobile

-------- Original Message --------

FormerLurker, 2022-04-26:
It would be interesting to see the configuration files you used to compile marlin.  There are a few settings  in there that are important:

TX_BUFFER_SIZE
RX_BUFFER_SIZE
MAX_CMD_SIZE
SERIAL_OVERRUN_PROTECTION