FormerLurker/ArcWelderPlugin #230: option to disable automatic convert / compression

TylonHH, 2022-04-26:
Hi,
I use the plugin [continuous print](https://plugins.octoprint.org/plugins/continuousprint/). The plugin generate gcodes to clean the bed.
In this moment your awesome plugin is converting the new gcode. Thats bad in this workflow.
So I have to disable your plugin within your menu. But then the manual compression in the file list also disappear.

So I request an option to disable the auto-compression. I still wanna use the button within the file-list.

I hope you understand what I mean.

kind regards

FormerLurker, 2022-04-26:
Is there not an 'auto process' option in your version?  It has honestly been a long time since I looked at the master release, so maybe it isn't there.

On a positive note, even if this isn't available in your version, there is a newer version you are welcome to try.  It is packed with all kinds of goodies.  Let me know!

TylonHH, 2022-04-26:
Shame on me: I'm on Arc-Welder (1.0.1rc1.dev2) because of the experimental feature "3D Arcs(vase mode)".
If the newest version has the vase mode too, I'll test it.

FormerLurker, 2022-04-26:
It does, as well as travel arc support and better statistics!  I just pushed new changes last weekend.

TylonHH, 2022-04-28:
> It does, as well as travel arc support and better statistics! I just pushed new changes last weekend.

The [latest release](https://github.com/FormerLurker/ArcWelderPlugin/releases) I see is 1.0.0 from 10/2020 and the latest RC from 11/2021
As I wrote I'm at 1.0.1rc1.dev2. I guess you mean this [RC](https://github.com/FormerLurker/ArcWelderPlugin/tree/rc/maintenance).

FormerLurker, 2022-04-28:
No, these changes aren't in that release.  Try installing with the following url in the plugin manager (get more-> From url...): ```https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/heads/rc/maintenance.zip```

TylonHH, 2022-04-29:
yeah I did. But there I still see no "auto process" option.

closed 2022-06-14 by TylonHH