FormerLurker/ArcWelderPlugin #232: Arc Welder and Cura 5 compability

luizbgomide, 2022-05-02:
Knowing that Cura 5 has variable line width, how does that affect Arc Welder?

Does it still work? Will it need an update to handle varying lines?

FormerLurker, 2022-05-02:
I have variable line width support in the latest branch.  Am hoping to push this build out soon.  Let me know if you would like to test it!

luizbgomide, 2022-05-02:
I could try it, but I'm not sure how to properly test if it is working properly besides noticing blobs and stutters (which I think it should solve completely). How would I know if the variable line width is working??

FormerLurker, 2022-05-02:
> How would I know if the variable line width is working??

Good question...  I'm honestly not sure, unfortunately.  I guess one way would be to make a comparison between prints, one without arcs and one with, but that's easier said than done.  If anyone has any suggestions, maybe drop a comment here.

luizbgomide, 2022-05-02:
Can your code identify varying line widths segments? Because if so, and the arcs are always generated from segments with the same width that is a non issue and could be diagnosed by verifying if the amount of filament extruded is roughly the same with or without arcs.

Otherwise I don't know how AW would possibly work, because if a single arc is generated from segments of multiple widths that arc will be wrong...

FormerLurker, 2022-05-02:
Before each segment is added to the shape detector, a flow rate is calculated.  AW has a new parameter to determine how much the flow rate can vary from point to point before an arc is rejected (5% is the default).  If the flow changes more than that, a new arc is started, preserving the line width.

Note that the flow rate varies by 5% or more even without the arachne engine, so I'm hopeful that the default is working.

Because of this, the total extrusion should be the same even if line widths aren't being taken into consideration, unfortunately.

luizbgomide, 2022-05-02:
Also regarding Cura and precision, what is the better approach/settings to setup a Cura + AW workflow to get the best representation of the model?

Does decreasing Maximum Resolution (default to 0.25) and/or Maximum Deviation ( 0.025) in Cura helps?

On a side note (while talking about precision), I think that a Maximum Arc Radius of 1 KM is too much and probably counterproductive. I had many straight segments turn into arcs with huge radii, that is probably harder on the MCU than a few long straight segments.

luizbgomide, 2022-05-25:
@FormerLurker I was using ArcWelder in Octoprint for the past weeks and it worked as expected (as far as I can tell).

Now, I just noticed that the ArcWelder Cura Plugin is working again, does that one uses the same code generation as the Octoprint version?

FormerLurker, 2022-06-07:
@luizbgomide, it is likely newer.  I've been trying to get that all updated, but have been having a lot of trouble treading water recently.

seantapscott, 2022-10-11:
Wouldn't a gcode previewer show whether or not the lines were printed at
the appropriate width?  Like, the model would look all screwy, layers
wouldn't exactly be on top of one another in the corners or curves if
ArcWelder ruined the variability of the line widths.