FormerLurker/ArcWelderPlugin #234: OrangePi-Zero & Arc-Welder

billbrach, 2022-05-16:
[image]

Hello, After hearing rave reviews of how good Arc-Welder is, I decided to add it to my OctoPrint install on a couple of OrangePI Zero's that I have attached to my printers.  One unit, I'll call it OPi-Green, it installed w/o issue.  When I looked at what verison of OP I was running on Green, I was surprised to see it was 1.8.0rc5.  OPi-Red was running the latest stable release 1.7.3.  Tried many things to get it running on Red, to no avail.  I have those error logs someplace; it always fails in the "wheel build" area of the code.  So, I upgraded Red to 1.8.0rc5, expecting it to install.

Here is the error message.  Note that this is a running OP install, with many prints under its belt, running 1.7.3

[image]

Running: Armbian 21.02.2 Buster with Linux 5.15.25-sunxi

FormerLurker, 2022-05-16:
What version of python are you using?  The devel branch may have a patch to fix this issue .

billbrach, 2022-05-16:
Python 3.7.3, pip 22.1

billbrach, 2022-05-16:
Here is the bottom of the log file, when I tried to install AW with OP 1.7.3  May be a compatibility problem with the Arm GCC compiler ??

   octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.h:413:35: warning:   ‘source_target_segment_statistics arc_welder::segment_statistics_’ [-Wreorder]
      source_target_segment_statistics segment_statistics_;
                                       ^~~~~~~~~~~~~~~~~~~
    octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.cpp:39:1: warning:   when initialized here [-Wreorder]
     arc_welder::arc_welder(std::string source_path, std::string target_path, logger * log, double resolution_mm, double max_radius, bool g90_g91_influences_extruder, int buffer_size, progress_callback callback) : current_arc_(DEFAULT_MIN_SEGMENTS, buffer_size - 5, resolution_mm, max_radius), segment_statistics_(segment_statistic_lengths, segment_statistic_lengths_count, log)
     ^~~~~~~~~~
    octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.cpp: In member function ‘arc_welder_results arc_welder::process()’:
    octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.cpp:284:16: warning: unused variable ‘end_clock’ [-Wunused-variable]
      const clock_t end_clock = clock();
                    ^~~~~~~~~
    octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.cpp: In member function ‘int arc_welder::process_gcode(parsed_command, bool, bool)’:
    octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.cpp:349:7: warning: unused variable ‘clear_shapes’ [-Wunused-variable]
      bool clear_shapes = false;
           ^~~~~~~~~~~~
    In file included from octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.h:38,
                     from octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.cpp:29:
    octoprint_arc_welder/data/lib/c/gcode_processor_lib/logger.h: At global scope:
    octoprint_arc_welder/data/lib/c/gcode_processor_lib/logger.h:37:20: warning: ‘log_level_names’ defined but not used [-Wunused-variable]
     static const char* log_level_names[] = {"NOSET", "VERBOSE", "DEBUG", "INFO", "WARNING", "ERROR", "CRITICAL"};
                        ^~~~~~~~~~~~~~~
    In file included from /usr/include/c++/8/vector:69,
                     from octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.h:28,
                     from octoprint_arc_welder/data/lib/c/arc_welder/arc_welder.cpp:29:
    /usr/include/c++/8/bits/vector.tcc: In member function ‘void std::vector<_Tp, _Alloc>::_M_realloc_insert(std::vector<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {segment_statistic}; _Tp = segment_statistic; _Alloc = std::allocator<segment_statistic>]’:
    /usr/include/c++/8/bits/vector.tcc:413:7: note: parameter passing for argument of type ‘std::vector<segment_statistic>::iterator’ {aka ‘__gnu_cxx::__normal_iterator<segment_statistic*, std::vector<segment_statistic> >’} changed in GCC 7.1
           vector<_Tp, _Alloc>::
           ^~~~~~~~~~~~~~~~~~~
    /usr/include/c++/8/bits/vector.tcc: In member function ‘arc_welder_progress arc_welder::get_progress_(long int, double)’:
    /usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<segment_statistic*, std::vector<segment_statistic> >’ changed in GCC 7.1
        _M_realloc_insert(end(), std::forward<_Args>(__args)...);
        ^~~~~~~~~~~~~~~~~
    /usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<segment_statistic*, std::vector<segment_statistic> >’ changed in GCC 7.1
        _M_realloc_insert(end(), std::forward<_Args>(__args)...);
        ^~~~~~~~~~~~~~~~~
    /usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<segment_statistic*, std::vector<segment_statistic> >’ changed in GCC 7.1
        _M_realloc_insert(end(), std::forward<_Args>(__args)...);
        ^~~~~~~~~~~~~~~~~
    /usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<segment_statistic*, std::vector<segment_statistic> >’ changed in GCC 7.1
        _M_realloc_insert(end(), std::forward<_Args>(__args)...);
        ^~~~~~~~~~~~~~~~~
    /usr/include/c++/8/bits/vector.tcc: In constructor ‘arc_welder::arc_welder(std::__cxx11::string, std::__cxx11::string, logger*, double, double, bool, int, progress_callback)’:
    /usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<segment_statistic*, std::vector<segment_statistic> >’ changed in GCC 7.1
        _M_realloc_insert(end(), std::forward<_Args>(__args)...);
        ^~~~~~~~~~~~~~~~~
    /usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<segment_statistic*, std::vector<segment_statistic> >’ changed in GCC 7.1
        _M_realloc_insert(end(), std::forward<_Args>(__args)...);
        ^~~~~~~~~~~~~~~~~
    /usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<segment_statistic*, std::vector<segment_statistic> >’ changed in GCC 7.1
        _M_realloc_insert(end(), std::forward<_Args>(__args)...);
        ^~~~~~~~~~~~~~~~~
    /usr/include/c++/8/bits/vector.tcc:109:4: note: parameter passing for argument of type ‘__gnu_cxx::__normal_iterator<segment_statistic*, std::vector<segment_statistic> >’ changed in GCC 7.1
        _M_realloc_insert(end(), std::forward<_Args>(__args)...);
[... cut ...] ead().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-un3956tx/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python3.7/Arc-Welder Check the logs for full command output.
WARNING: You are using pip version 20.3.3; however, version 22.1 is available.
You should consider upgrading via the '/home/pi/oprint/bin/python3 -m pip install --upgrade pip' command.

ldursw, 2022-05-16:
```
arm-linux-gnueabihf-gcc: fatal error: Killed signal terminated program cc1plus
compilation terminated.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```

It seems the board doesn't have enough RAM to compile everything. Enabling swap may work but will be extremely slow.

FormerLurker, 2022-05-16:
Interesting.  Maybe it could be possible to precompile the lib, but the install script would need some modifications.  Have you considered the console version?  It can be integrated into most slicers.  There is a cura plugin too.

FormerLurker, 2022-05-17:
I will try installing on that rc ASAP.  Certainly could be an issue.

billbrach, 2022-05-17:
> ```
> arm-linux-gnueabihf-gcc: fatal error: Killed signal terminated program cc1plus
> compilation terminated.
> error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
> ```
> 
> It seems the board doesn't have enough RAM to compile everything. Enabling swap may work but will be extremely slow.

You may be correct !! The one that I can install AW on has 512M RAM and the one I can't is only 256M.

billbrach, 2022-05-17:
Problem solve - it was the RAM issue as mentioned above.  What I did was on Green uninstall AW and roll back OP to the current stable, 1.7.3.  Then installed AW, without any issues !!  So, 256M is not enough to install AW.  I'm going to swap SD cards, install AW on Red's SD card using Green, and then reinstall the SD card into Red.  Closing the issue....

billbrach, 2022-05-17:
Closed, thanks for all of your help !!

closed 2022-05-17 by billbrach

FormerLurker, 2022-05-18:
Genius!  I bet it will too!  I will have to update the readme!

FormerLurker, 2022-06-07:
@billbrach, did that work?

billbrach, 2022-06-07:
FormerLurker - YES, swapping the SD card into the unit with more RAM did allow it to compile and install.

NO - when I put that SD card back into the lower RAM OPi, Arc Welder does not show up in the list of installed plugins.  After I installed AW in the unit with higher RAM, I uploaded a file, and saw that AW did its magic, so I know it installed correctly.

FYI - the OPi that won't compile or run, is 256M RAM, the other unit that works fine is 512M RAM.

Thanks for your help, and thanks for a great plugin !!