FormerLurker/ArcWelderPlugin #235: Updater not working under OctoPrint v1.8.0

Taomyn, 2022-05-18:
I have updated to OctoPrint v1.8.0 (RPi 4) and everything looks fine except now when I perform a software update check it fails for ArcWelder:

[image]

```
2022-05-18 06:49:06,867 - octoprint.plugins.softwareupdate - ERROR - Could not check arc_welder for updates
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1835, in _get_current_version
    target, check, online=online, credentials=credentials
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/version_checks/github_release.py", line 276, in get_latest
    target
octoprint.plugins.softwareupdate.exceptions.ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None
2022-05-18 06:49:07,039 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2022-05-18 06:49:12,510 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2022-05-18 06:49:12,711 - octoprint.plugins.softwareupdate - ERROR - Could not check arc_welder for updates
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1835, in _get_current_version
    target, check, online=online, credentials=credentials
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate/version_checks/github_release.py", line 276, in get_latest
    target
octoprint.plugins.softwareupdate.exceptions.ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None

```

I have tried all the release channels and they do the same, although the plug-in itself does seem to be still working for now.

FormerLurker, 2022-06-07:
Yeah, this has been broken since the beginning due to a typo in the plugin definition.  I need to submit a pull request for Octoprint to fix existing installs.  V sorry about that!

Taomyn, 2022-10-31:
Has this been abandoned?

Taomyn, 2022-11-03:
@FormerLurker no worries, no other issues with the plugin, just spotted when looking at another problem with my OctoPrint updates and thought I'd check here.

jimmiedave, 2023-01-09:
Yeah, troubling that plugin check-for-update won't work. Sounds like a simple fix (always "sounds like...", I know): just provide the right release and repo and user info, IIRC.

greatflyingsteve, 2024-12-17:
@FormerLurker Is the PR just a fix for the plugin's metadata?  Is it a thing one of us might be able to help with?