FormerLurker/ArcWelderPlugin #239: Installation error: `ModuleNotFoundError: No module named 'distutils.msvccompiler'`

MisterEyMan, 2022-08-16:
I have Octoprint setup on my DietPi.  When trying to install Arc Welder I get:
```
root@DietPi:~# pip install "https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip"
Collecting https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip
  Downloading https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip
     / 200.1 kB 657.6 kB/s 0:00:00
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-req-build-0xt8qnjh/setup.py", line 32, in <module>
          from distutils.msvccompiler import MSVCCompiler
      ModuleNotFoundError: No module named 'distutils.msvccompiler'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
```

Any ideas?

kamilko, 2024-12-04:
This issue still exists in setup DietPi+OctoPrint from software manager. Everything installed fresh on 2024.12.01. on raspberry pi.

gisforgirard, 2025-04-16:
https://github.com/pypa/setuptools/issues/4620#issuecomment-2323233905