FormerLurker/ArcWelderPlugin #257: Cannot install from OctoPrint or command line: ModuleNotFoundError: No module named 'octoprint'

aaronsnoswell, 2023-05-01:
I'm trying to install ArcWelder from OctoPrint's interface (and also from SSH command line). The install fails with `ModuleFoundError: No module named 'octoprint'`

```bash
pi@tiltpi:~ $ cd OctoPrint/venv/bin/
pi@tiltpi:~/OctoPrint/venv/bin $ source activate
(venv) pi@tiltpi:~/OctoPrint/venv/bin $ ./pip install "https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip"
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip
  Using cached https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [18 lines of output]
      Unknown Version, falling back to 1.0.0+u.bb71e8f.
      Traceback (most recent call last):
        File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-0ora9c37/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 341, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
        File "/tmp/pip-build-env-0ora9c37/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 323, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-0ora9c37/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 488, in run_setup
          self).run_setup(setup_script=setup_script)
        File "/tmp/pip-build-env-0ora9c37/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 338, in run_setup
          exec(code, locals())
        File "<string>", line 84, in <module>
      ModuleNotFoundError: No module named 'octoprint'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

The same error was reported when trying to install from the OctoPrint web interface.

Let me know what debugging information would be useful to have. My Raspberry Pi is a Model B Rev 2. For now, I'm using the command line ArcWelder binary manually after slicing, but it would be lovely to get it this working in OctoPrint :)

Thanks for an awesome piece of software!

aaronsnoswell, 2023-05-01:
Update... after browsing the issues database, I tried installing the `devel` branch and got the same error;

```bash
(venv) pi@tiltpi:~/OctoPrint/venv/bin $ ./pip install https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip
  Downloading https://github.com/FormerLurker/ArcWelderPlugin/archive/devel.zip
     | 288.8 kB 129.6 kB/s 0:00:03
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [18 lines of output]
      Unknown Version, falling back to 1.1.0rc3+u.2cfe877.
      Traceback (most recent call last):
        File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/home/pi/OctoPrint/venv/lib/python3.7/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-7j3j3koz/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 341, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=['wheel'])
        File "/tmp/pip-build-env-7j3j3koz/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 323, in _get_build_requires
          self.run_setup()
        File "/tmp/pip-build-env-7j3j3koz/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 488, in run_setup
          self).run_setup(setup_script=setup_script)
        File "/tmp/pip-build-env-7j3j3koz/overlay/lib/python3.7/site-packages/setuptools/build_meta.py", line 338, in run_setup
          exec(code, locals())
        File "<string>", line 84, in <module>
      ModuleNotFoundError: No module named 'octoprint'
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

aaronsnoswell, 2023-05-01:
Update: seems possibly related to: https://github.com/FormerLurker/Octolapse/issues/576

scubix, 2023-05-04:
I have the exact same problem and can not get anything installed on a fresh installation of ubuntu / octoprint

aaronsnoswell, 2023-05-06:
Yup. I've just set up a fresh install on Ubuntu instead of raspbian (nothing else installed on the system!) as well and it's got the same issue.

aaronsnoswell, 2023-05-06:
I also can't install any other plugins though, so maybe it's not an arcwelder thing? Maybe OctoPrint in a venv is broken in recent updates?

aaronsnoswell, 2023-05-06:
Seems it's not just us: https://community.octoprint.org/t/unable-to-install-any-octoprint-plugins/51644/2

aaronsnoswell, 2023-05-06:
Turns out `~/OctoPrint/venv/bin/pip install wheel` fixes this!

closed 2023-05-06 by aaronsnoswell