FormerLurker/ArcWelderPlugin #268: Issues Installing

charely6, 2024-05-19:
I just setup a new install of octoprint using octoprint_deploy and I'm having issues getting ArcWelder installed.

I cleared the plugin manager log file and then retried the install to get a fresh log of what I'm getting. it looks like its parsing it weird and getting stuck on some character but I'm not sure what's causing that.

2024-05-19 15:09:54,507   /home/charely6/OctoPrint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp8wl0vnoc/ArcWelderPlugin-master.zip --no-cache-dir
2024-05-19 15:09:58,009 > Processing /tmp/tmp8wl0vnoc/ArcWelderPlugin-master.zip
2024-05-19 15:09:58,010 > Preparing metadata (setup.py): started
2024-05-19 15:09:58,011 > Preparing metadata (setup.py): finished with status 'error'
2024-05-19 15:09:58,015 ! error: subprocess-exited-with-error
2024-05-19 15:09:58,016 ! 
2024-05-19 15:09:58,016 ! × python setup.py egg_info did not run successfully.
2024-05-19 15:09:58,017 ! │ exit code: 1
2024-05-19 15:09:58,017 ! ╰─> [18 lines of output]
2024-05-19 15:09:58,017 ! /tmp/pip-req-build-3v9kxr55/versioneer.py:421: SyntaxWarning: invalid escape sequence '\s'
2024-05-19 15:09:58,017 ! LONG_VERSION_PY['git'] = '''
2024-05-19 15:09:58,018 ! Traceback (most recent call last):
2024-05-19 15:09:58,018 ! File "<string>", line 2, in <module>
2024-05-19 15:09:58,018 ! File "<pip-setuptools-caller>", line 34, in <module>
2024-05-19 15:09:58,018 ! File "/tmp/pip-req-build-3v9kxr55/setup.py", line 53, in <module>
2024-05-19 15:09:58,019 ! plugin_version = versioneer.get_version()
2024-05-19 15:09:58,019 ! ^^^^^^^^^^^^^^^^^^^^^^^^
2024-05-19 15:09:58,019 ! File "/tmp/pip-req-build-3v9kxr55/versioneer.py", line 1480, in get_version
2024-05-19 15:09:58,019 ! return get_versions()["version"]
2024-05-19 15:09:58,020 ! ^^^^^^^^^^^^^^
2024-05-19 15:09:58,020 ! File "/tmp/pip-req-build-3v9kxr55/versioneer.py", line 1412, in get_versions
2024-05-19 15:09:58,020 ! cfg = get_config_from_root(root)
2024-05-19 15:09:58,020 ! ^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-05-19 15:09:58,021 ! File "/tmp/pip-req-build-3v9kxr55/versioneer.py", line 342, in get_config_from_root
2024-05-19 15:09:58,021 ! parser = configparser.SafeConfigParser()
2024-05-19 15:09:58,021 ! ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
2024-05-19 15:09:58,021 ! AttributeError: module 'configparser' has no attribute 'SafeConfigParser'. Did you mean: 'RawConfigParser'?
2024-05-19 15:09:58,022 ! [end of output]
2024-05-19 15:09:58,022 ! 
2024-05-19 15:09:58,022 ! note: This error originates from a subprocess, and is likely not a problem with pip.
2024-05-19 15:09:58,022 ! error: metadata-generation-failed
2024-05-19 15:09:58,023 ! 
2024-05-19 15:09:58,023 ! × Encountered error while generating package metadata.
2024-05-19 15:09:58,023 ! ╰─> See above for output.
2024-05-19 15:09:58,023 ! 
2024-05-19 15:09:58,024 ! note: This is an issue with the package mentioned above, not pip.
2024-05-19 15:09:58,024 ! hint: See above for details.

R1D3R175, 2024-07-11:
Having the same issue, can't install

R1D3R175, 2024-07-11:
Could be related to pygeos/pygeos#463

gisforgirard, 2025-04-16:
https://github.com/FormerLurker/ArcWelderPlugin/compare/master...gisforgirard:ArcWelderPlugin:master