FormerLurker/ArcWelderPlugin #270: Issue Install after update Ubuntu to the latest version

XeroZar, 2024-10-28:
I recently did a clean install on my Ubuntu system. All other plugins install successfully except Arc Welder... Apparently its not compatibale with the latest Python... Please help.

Installing plugin "Arc Welder" from https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip...
/home/pi/OctoPrint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmprpiwt1_5/ArcWelderPlugin-master.zip --no-cache-dir
Processing /tmp/tmprpiwt1_5/ArcWelderPlugin-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'error'
error: subprocess-exited-with-error

× python setup.py egg_info did not run successfully.
│ exit code: 1
╰─> [6 lines of output]
Traceback (most recent call last):
File "<string>", line 2, in <module>
File "<pip-setuptools-caller>", line 34, in <module>
File "/tmp/pip-req-build-54ma7jnj/setup.py", line 32, in <module>
from distutils.msvccompiler import MSVCCompiler
ModuleNotFoundError: No module named 'distutils.msvccompiler'
[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

ninjamojo, 2024-11-11:
I will take a look at this ASAP, probably over thanksgiving break.  Thanks for the heads up.

fiechr, 2024-11-24:
Same issue here after upgrading to the latest Octoprint version on Ubuntu 24.04.1 LTS (noble).

pebdev, 2025-01-09:
I found a workaround for all Linux architectures (tested on Dietpi for me).
I provide you a [master-patched.zip](https://github.com/user-attachments/files/18367934/master-patched.zip). I applied these changes : 

### 1- Download master source code
`wget https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip`

### 2- Unzip archive
```
unzip master.zip -d master
cd master
```

### 3- Edit setup file
`vim ArcWelderPlugin-master/setup.py`

_Note : I had an issue by using Visual Studio Code at the end in Octoprint during the installation step (maybe a mode issue). I didn't wast any time with this, but be careful at this step._

Comment all of these lines  : 
```
    #from distutils.msvccompiler import MSVCCompiler
    #from distutils.bcppcompiler import BCPPCompiler
    #from distutils.cygwinccompiler import CygwinCCompiler
```

```
    #MSVCCompiler.compiler_type: {
    #    "extra_compile_args": ["/O2", "/GL", "/Gy", "/MD", "/EHsc"],
    #    "extra_link_args": [],
    #    "define_macros": [],
    #},

    ...

    #BCPPCompiler.compiler_type: {
    #    "extra_compile_args": ["-O3", "-std=c++11"],
    #    "extra_link_args": [],
    #    "define_macros": [],
    #},
    #CygwinCCompiler.compiler_type: {
    #    "extra_compile_args": ["-O3", "-std=c++11"],
    #    "extra_link_args": [],
    #    "define_macros": [],
    #},
```

```
    #MSVCCompiler.compiler_type: {
    #    "extra_compile_args": ["/EHsc", "/Z7"],
    #    "extra_link_args": ["/DEBUG"],
    #    "define_macros": [],
    #},

    ...

    #BCPPCompiler.compiler_type: {
    #    "extra_compile_args": [],
    #    "extra_link_args": [],
    #    "define_macros": [],
    #},
    #CygwinCCompiler.compiler_type: {
    #    "extra_compile_args": [],
    #    "extra_link_args": [],
    #    "define_macros": [],
    #},

```
### 4- Re-create the zip file
`zip -r master-patched.zip .`

### 5- Install plugin
In Octoprint, install plugin from this archive.

gisforgirard, 2025-04-16:
https://github.com/gisforgirard/ArcWelderPlugin

gisforgirard, 2025-04-16:
> msvccompiler

https://github.com/pypa/setuptools/issues/4620#issuecomment-2323233905

DanielmGM, 2025-08-09:
I've figured out! I'm running Ubuntu 24.04
I've based on @pebdev 's answer and somehow 

`wget https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip
unzip master.zip -d master
cd master/ArcWelderPlugin-master

# change SafeConfigParser to ConfigParser
sed -i 's/SafeConfigParser/ConfigParser/g' versioneer.py

#change readfp() to read_file()
sed -i 's/\.readfp(/.read_file(/g' versioneer.py

#delete imports of MSVC/BCPP/Cygwin on setup.py
sed -i 's/from distutils.core import setup/from setuptools import setup/' setup.py
sed -i '/from distutils.msvccompiler import MSVCCompiler/ s/^/#/' setup.py
sed -i '/from distutils.bcppcompiler import BCPPCompiler/ s/^/#/' setup.py
sed -i '/from distutils.cygwinccompiler import CygwinCCompiler/ s/^/#/' setup.py
sed -i '/MSVCCompiler.compiler_type:/,/},/ s/^/#/' setup.py
sed -i '/BCPPCompiler.compiler_type:/,/},/ s/^/#/' setup.py
sed -i '/CygwinCCompiler.compiler_type:/,/},/ s/^/#/' setup.py

cd ../..
zip -r master-patched.zip master/ArcWelderPlugin-master
`
I don't know if there's anything we can do to fix the master with those changes

XeroZar, 2025-08-16:
> I've figured out! I'm running Ubuntu 24.04 I've based on [@pebdev](https://github.com/pebdev) 's answer, made some changes for compatibility and somehow it worked
> 
> `wget https://github.com/FormerLurker/ArcWelderPlugin/archive/master.zip unzip master.zip -d master cd master/ArcWelderPlugin-master
> # change SafeConfigParser to ConfigParser
> 
> sed -i 's/SafeConfigParser/ConfigParser/g' versioneer.py
> 
> #change readfp() to read_file() sed -i 's/.readfp(/.read_file(/g' versioneer.py
> 
> #delete imports of MSVC/BCPP/Cygwin on setup.py sed -i 's/from distutils.core import setup/from setuptools import setup/' setup.py sed -i '/from distutils.msvccompiler import MSVCCompiler/ s/^/#/' setup.py sed -i '/from distutils.bcppcompiler import BCPPCompiler/ s/^/#/' setup.py sed -i '/from distutils.cygwinccompiler import CygwinCCompiler/ s/^/#/' setup.py sed -i '/MSVCCompiler.compiler_type:/,/},/ s/^/#/' setup.py sed -i '/BCPPCompiler.compiler_type:/,/},/ s/^/#/' setup.py sed -i '/CygwinCCompiler.compiler_type:/,/},/ s/^/#/' setup.py
> 
> cd ../.. zip -r master-patched.zip master/ArcWelderPlugin-master ` I don't know if there's anything we can do to fix the master with those changes

I tried this still didn't work... please see attached log.

[plugin_pluginmanager_console.log](https://github.com/user-attachments/files/21815488/plugin_pluginmanager_console.log)